iMicknl/LoctekMotion_IoT #6: None of the commands work.

staxDB, 2021-03-28:
Hey, damn cool project and just what I was looking for!

I am having a few issues in my setup though: I have been trying ,to control a flexispot desk remotely. Unfortunately none of the scenarios work. 
I first tried the ESPHome solution and used an ESP8266. Since this did not work (or my desk did not respond at all), I still tried the variant with the Raspberry Pi - Unfortunately also unsuccessful. I even tried to execute the single "Up" command alone with a Python script, but the desk does not react to it.

Which setup I have: A Flexispot E5B with an HS01B-1 control unit and a CB38M2A-1 control module. I connected the Raspberry Pi and also the ESP8266 to the control module (CB38M2A-1) via a modified Ethernet cable. For the correct ethernet cable mapping I used this table as a guideline:
https://github.com/deskmatik/deskmatik#wiring-flexispot

Do you have any idea where I made a mistake? Is my control module possibly not compatible, since I also have only one Ethernet port?

iMicknl, 2021-03-28:
Have you tried reversing the TX / RX cables? This is very messy always and I had those reversed many many times during the development. The PIN20 is a very important one as well, since this allows you to send commands. 

Could you perhaps share more here about how you configured the cables and how you set up the ESPHome yaml?

The Raspberry Pi code could use some love by the way, I just used this during the testing. The ESPHome code should work and this is what I am currently using.

Happy to walk through it on Discord as well, iMick#1903.

lordfiSh, 2021-03-31:
Same Problem here, also tried to switch RX/TX

[image]

```
substitutions:
  device_name: Flexispot EK5
  name: flexispot_ek5
  min_height: "73.6" # Min height + 0.1
  max_height: "122.9" # Max height - 0.1

esphome:
  name: ${name}
  comment: ${device_name}
  platform: ESP8266 # TODO Change to your platform
  board: nodemcuv2 # TODO Change to your board
  includes:
    - desk_height_sensor.h

wifi:
  ssid: !secret esphome_wlanssid
  password: !secret esphome_wlanpw 

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${device_name} Fallback Hotspot"
    password: !secret esphome_web_pass

captive_portal:

# Enable logging
logger:
  #level: DEBUG
  baud_rate: 0

# Enable Home Assistant API
api:
  password: !secret esphome_ota

ota:
  password: !secret esphome_ota

uart:
  id: desk_uart
  baud_rate: 9600
  tx_pin: D7
  rx_pin: D8

sensor:
  - platform: wifi_signal
    name: "WiFi Signal"
    update_interval: 60s

  - platform: uptime
    name: Uptime

  - platform: custom
    lambda: |-
      auto desk_height_sensor = new DeskHeightSensor(id(desk_uart));
      App.register_component(desk_height_sensor);
      return {desk_height_sensor};
    sensors:
      id: "desk_height"
      name: Desk Height
      unit_of_measurement: cm
      accuracy_decimals: 1
      icon: "mdi:counter"

switch:
  - platform: gpio
    name: "Virtual Screen"
    pin:
      number: D5
      mode: OUTPUT
    restore_mode: ALWAYS_OFF
    internal: true

  - platform: uart
    name: "Preset 1"
    id: switch_preset1
    icon: mdi:numeric-1-box
    data: [0x9b, 0x06, 0x02, 0x04, 0x00, 0xac, 0xa3, 0x9d]
    uart_id: desk_uart

  - platform: uart
    name: "Preset 2"
    id: switch_preset2
    icon: mdi:numeric-2-box
    data: [0x9b, 0x06, 0x02, 0x08, 0x00, 0xac, 0xa6, 0x9d]
    uart_id: desk_uart

  - platform: uart
    name: "Sit" # Preset 3 on some control panels
    id: switch_sit
    icon: mdi:chair-rolling
    data: [0x9b, 0x06, 0x02, 0x00, 0x01, 0xac, 0x60, 0x9d]
    uart_id: desk_uart

  - platform: uart
    name: "Stand" # Not available for all control panels
    id: switch_stand
    icon: mdi:human-handsup
    data: [0x9b, 0x06, 0x02, 0x10, 0x00, 0xac, 0xac, 0x9d]
    uart_id: desk_uart

  - platform: uart
    name: "Up"
    id: switch_up
    icon: mdi:arrow-up-bold
    data: [0x9b, 0x06, 0x02, 0x01, 0x00, 0xfc, 0xa0, 0x9d]
    uart_id: desk_uart
    internal: true

  - platform: uart
    name: "Down"
    id: switch_down
    icon: mdi:arrow-down-bold
    data: [0x9b, 0x06, 0x02, 0x02, 0x00, 0x0c, 0xa0, 0x9d]
    uart_id: desk_uart
    internal: true

  - platform: uart
    name: "M"
    id: switch_m
    icon: mdi:alpha-m-circle
    data: [0x9b, 0x06, 0x02, 0x20, 0x00, 0xac, 0xb8, 0x9d]
    uart_id: desk_uart

  - platform: uart
    name: "(wake up)" # Not available on all control panels
    id: switch_wake_up
    icon: mdi:gesture-tap-button
    data: [0x9b, 0x06, 0x02, 0x00, 0x00, 0x6c, 0xa1, 0x9d]
    uart_id: desk_uart

cover:
  - platform: template
    # icon: mdi:table-chair
    # icon: mdi-human-male-height-variant
    name: "Desk"
    assumed_state: true

    # Move desk up
    open_action:
      - while:
          condition:
            sensor.in_range:
              id: desk_height
              below: ${max_height}
          then:
          - logger.log: "Executing up command"
          - switch.turn_on: switch_up
          - delay: 10ms
    
    # Move desk down
    close_action:
      - while:
          condition:
            sensor.in_range:
              id: desk_height
              above: ${min_height} 
          then:
          - logger.log: "Executing down command"
          - switch.turn_on: switch_down
          - delay: 10ms
    optimistic: true
```

iMicknl, 2021-03-31:
@staxDB I just discussed this issue briefly with @lordfiSh and he is indeed facing similar issues. Looking at DeskMatic, it seems that they set the PIN20 to HIGH, where I use LOW.

Perhaps the following works;

```
  - platform: gpio
    name: "Virtual Screen"
    pin:
      number: D2
      mode: OUTPUT
    restore_mode: ALWAYS_OFF
    internal: true
```

Set restore_mode to `ALWAYS_ON`, or remove the `internal` and toggle it yourself during testing.

staxDB, 2021-04-04:
I just tried the customization: It seems to work! However, I have also changed the PIN assignment: 
PIN20 = D5
RX = D7
TX = D8

closed 2021-04-04 by staxDB

iMicknl, 2021-04-04:
@staxDB are you able to do a pull request on this repo to add better instructions for HS01B-1? I will investigate the `ALWAYS_ON`/ `ALWAYS_OFF` differences between our model and document that in a better way.

iMicknl, 2021-04-05:
Should be fixed with [new ESPHome sample](https://github.com/iMicknl/LoctekMotion_IoT/tree/main/packages/esphome), thanks @staxDB.