iMicknl/LoctekMotion_IoT #7: Improvement: Still be able to use the control panels

staxDB, 2021-04-04:
It would be awesome if you could still use the control panel. The ESP8266 would have to read the incoming signals from the control panel and pass them on to the control unit. The same applies the other way round: Everything that is read from the control unit should also be written back to the control panel. Would such a setup be possible?

| RJ45 PIN | Original cable color | Ethernet cable color | Description    | ESP8266 GPIO > Control Unit | ESP8266 GPIO > Control Panel | Control Panel  |
|----------|----------------------|----------------------|----------------|-----------------------------|------------------------------|----------------|
| 1        | brown                | white-orange         | RES            |                             |                              | RES            |
| 2        | white                | orange               | SWIM           |                             |                              | SWIM           |
| 3        | purple               | white-green          | N/C            |                             |                              | N/C            |
| 4        | red                  | blue                 | Enable / Awake | D2                          | ?                            | Enable / Awake |
| 5        | green                | white-blue           | RX (of remote) | D6                          | D1                           | TX (of remote) |
| 6        | black                | green                | TX (of remote) | D5                          | D0                           | RX (of remote) |
| 7        | blue                 | white-brown          | GND            |                             |                              | GND            |
| 8        | yellow               | brown                | VDD (5V)       |                             |                              | VDD (5V)       |

The ESP8266 itself could be supplied with power via e.g. USB, so that the control panel could use the power supply from the control unit.

iMicknl, 2021-04-04:
Have you tried a simple RJ45 splitter? I wonder if that would work as well.. 

Otherwise; it would be interesting to investigate if you can pass through all communication indeed.

staxDB, 2021-04-05:
I will give it a try

NazarLysyi, 2021-05-16:
Are you able to connect the panel to the ESP and listen command?

dasfuu, 2021-07-20:
Isn't that already possible? My desk seems to have two controller ports. Or is that only on specific models?

iMicknl, 2021-07-20:
@dasfuu my desk also has two controller ports, but unfortunately there are some models with just one controller port. That's the same reason why I haven't spend any time on researching this.

Diddlik, 2021-07-21:
@staxDB Do you have some progress on this?

lordfiSh, 2021-07-21:
> Isn't that already possible? My desk seems to have two controller ports. Or is that only on specific models?
@NazarLysyi My [Flexispot E5](https://www.amazon.de/-/en/gp/product/B077JPC7PK/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) has only one Port, Controller Model CB38M2A-1

[image]

pepsinio, 2021-08-02:
Anyone has success with this? Alternatively i guess an external keypad connected to other GPIO's can do the trick too (minus the display)

iMicknl, 2021-08-02:
I haven't heard it yet, however feel free to [join us on Discord](https://discord.gg/C7TNzUZ9Xf) to discuss further. I don't think that we have users there yet who only have a single input, however perhaps they have an idea :-).

pepsinio, 2021-08-11:
@iMicknl you can close this issue with my pull request as well

closed 2021-08-17 by iMicknl