iMicknl/LoctekMotion_IoT #9: Wakeup Issues with Esphome

Algram, 2021-06-05:
Hi there,

I've successfully integrated my flexispot desk to Home Assistant using your project. Thanks for that!

One problem I am having is, that I need to trigger the wakeup switch (and wait like 5 seconds) before I can send presets events.
Is there any way that this could be more integrated? Ideally I would like to just trigger the preset and the desk to always react.

Thanks!

iMicknl, 2021-06-06:
Great to hear! Which desk do you own?

Perhaps you could execute the wake up switch on `switch.turn_on`, however I am not sure if this will be the right order. https://esphome.io/components/switch/index.html#switch-turn-on-action

Algram, 2021-06-07:
@iMicknl  I am using the Flexispot E07 with the HS01B-1 control panel.

I thought about sending the wakeup event before the actual event, but there seems to be a big delay between the wakeup event and the other switches working. I also have to press the wakeup button multiple tims which seems weird.

N1c093, 2021-06-08:
@Algram Do you shut down your complete desk and ESP everyday?

I do so and implement this workaround for me, which seems to work fine. I send the button "M" to my table after the ESP is booted. After this the ESP can control the table all over the time.

I'm using "Flexispot EK2" with "HS13A-1" Control Panel and "CB38M2B(IB)-1" Control Box

```yaml

esphome:
  name: ${name}
  comment: ${device_name}
  platform: ESP8266 # TODO Change to your platform
  board: nodemcuv2 # TODO Change to your board
  includes:
    - desk_height_sensor.h
  on_boot:
    priority: -10
    then:
      - switch.turn_on: switch_m

```

Maybe this will also resolve your issue?

XPStef, 2022-02-22:
Hi there,

I own a Flexispot E7 as well and have deployed ESPHome on a D1 Mini Pro. Haven't disassembled the control panel but checked the wire output of the second RJ45 port and it matches the HS01B-1 pin layout (it has a touchscreen, though). At the moment using the built in web server for testing purposes.

What I've noticed is that the wake up toggle doesn't work for me at all (even after pressing it multiple times, trying to send e.g. the UP command afterwards, no luck...). Either the control panel has to be woken up first (displaying the current height) or the PIN20 (of the internal Virtual Screen) has to be set to HIGH/ON (restore_mode is ALWAYS_OFF on boot) to be able to send a command to the desk. Afterwards it works like expected.

Nevertheless, after waking up the display of the control panel or setting PIN20 to HIGH and sending a command to the desk, the control panel executes button presses differently. By default, when pressing a button, the display turns on first (displaying the current height) and after a delay of 1-2 seconds executes the command of the pressed button. After controlling the desk via the D1 Mini, the button presses on the control panel get executed immediately. This behavior stays like that until the desk gets rebooted/reconnected even after disconnecting the D1 Mini beforehand.

There is something different when using the control panel by itself, getting/displaying the current height of the table and then executing the button press and using ESPHome because it changes the behavior of the control panel or changes something in the control box that button presses get executed immediately.

Does anybody have a solution to circumvent this behavior?

@Algram: Have you found a solution for your problem? Maybe I (and others) can benefit from it as well :)

Thank you @iMicknl for the great project 👍

Algram, 2022-02-22:
@XPStef I still have the same issue and also have not found a way to fix it yet. So I am really interested in what you'll find too!

iMicknl, 2022-02-22:
> When waking up the display (displaying the current height), PIN20 receives a voltage of ~2.8V. When the display turns of, no voltage is applied anymore. Testing this by sending the wake up command, the voltage stays at 0V. So I guess the control box of the E7 isn't compatible with the wake up command?

I don't think the wake up command works for anyone, I took this from another project, but couldn't get it working! Personally what I do is call the "M" button on wake up, so I always get the current height. And than after every command (on control panel or ESP) it will get the new height as well.

By the way, feel free to do a PR to add your findings / hints to our README. As you can see, we can use some better documentation. :)

Algram, 2022-02-22:
@iMicknl Even if I toggle the M button, UP/Down and Preset commands after that do not work. I have to press them around 10 times for them to get registered. I have no idea where this is coming from, but I think @XPStef is having similar problems.

XPStef, 2022-02-22:
> So I am really interested in what you'll find too!
> 
> If you need me to test anything, just let me know, I'm down.

Not much unfortunately. But thanks for offering your help ;)

> I don't think the wake up command works for anyone, I took this from another project, but couldn't get it working! Personally what I do is call the "M" button on wake up, so I always get the current [...]

I've tried the 'M' switch 'workaround'. Works indeed but PIN20 has still to be set to HIGH/ON. If it's set to LOW/OFF, executing the 'switch.turn_on: switch_m' command on boot doesn't work, unfortunately.

@Algram: Try to set the 'restore_mode' of the 'Virtual Screen' to 'ALWAYS_ON' to get rid of the multiple wake ups. The current height can then be obtained by switching on the 'M' switch on boot as @N1c093 and @iMicknl has mentioned. I could just verify it on my desk.

Furthermore, I think I got the H13A-01 instead of the HS01B-1 as mentioned in Issue #23. They got an E7 and E8 as well with the touchscreen interface.

> UPDATE: When waking up the display (displaying the current height), PIN20 receives a voltage of ~2.8V. When the display turns of, no voltage is applied anymore. Testing this by sending the wake up command, the voltage stays at 0V. So I guess the control box of the E7 isn't compatible with the wake up command?

I also noticed, that once PIN20 was set to HIGH from the D1 Mini, the control box starts to permanently deliver a voltage of ~2.8V on PIN20 even after the display turns off (it should then be 0V again) until the desk gets rebooted/reconnected. Weird stuff tbh.
I think there is another command which gets sent from the H13A-01 to the control box (getting the current height) before the respective command gets executed. By simply touching one of the buttons the display turns on (wakes up), on the second touch the respective command (of any pressed/touched button) gets executed. There has to be another command to 'wake up' the control panel/box and obtain the current height of the desk (in my opinion). Like: IF display is OFF: WAKEUP; ELSE EXECUTE command ...

Actually I wouldn't bother at all. Everything works (for me) like it should. But with the touchscreen control panel there is no protection against accidental touches anymore because the first touch to turn on the display or the delay of 1-2 seconds before executing the button press/touch is gone once the famous PIN20 got set to HIGH.

We'll see. Maybe someone is more practical in finding the right command/PIN/whatever :)