iMicknl/LoctekMotion_IoT #10: [bug] HS13A-1 stuck on "LOC"

tsheaff, 2021-08-01:
Not sure this is a bug with this library, but I was playing around with connecting my arduino to the standing desk, and now I've notices my `HS13A-1` keypad is stuck in the "LOC" state. Not 100% sure how to reproduce, but it will not get out of this setting regardless of how I press the buttons (hold down M for a long time, hold down up and down for a long time, and many other combinations) — after all of these it still just says "LOC" and will not move the desk up and down. I'm not sure how to debug whether it's an issue with the control panel itself, or with the box it connects to. Is any state stored in this box? My box model is `CB38M2B(IB)-1`

Has anyone else run into this issue? Any ideas how to resolve? Do I need to just order a new HS13A-1 control panel? I had a hard time finding how to order one directly from LoctekMotion's website.  cc @iMicknl @staxDB any pointers would be amazing.

iMicknl, 2021-08-02:
Which code did you use? 

I would start with the following steps first:

1. Disconnect your Arduino from the motor control box
2. Remove the power cable of your desk
3. Wait 60 seconds
4. Plug in the power cable of your desk
5. Did this solve your problem?

I have exactly the same control box and control panel, however I am using it with ESPHome + Home Assistant at the moment.

By the way, there are some new users figuring out how it works, thus they created a Discord server. Feel free to [join us on Discord](https://discord.gg/C7TNzUZ9Xf).

mad2xlc, 2021-12-18:
LOC means lock and can be triggered with a 5 seconds press on memory and also released the same way.
It's not a bug it's a feature :)
I haven't attached my desk to an esp but came across this issue and was worried about this bug. But i think everything is fine here.

@tsheaff 
Were you able to solve the problem already?

iMicknl, 2022-01-05:
Thanks for the context @mad2xlc! Perhaps we should add this to the FAQ as well. 

I will close this issue for now.

closed 2022-01-05 by iMicknl

tsheaff, 2022-03-11:
Hey @iMicknl @mad2xlc apologies for the incredibly slow reply 😆 

I have not managed to fix this issue with the above. Holding down the memory for 5 seconds definitely doesn't fix it, or I would've fixed this months ago. I'm thinking of just buying an new controller. Any thoughts on the best place to purchase one?

RinkMax, 2022-07-14:
Hi @tsheaff,
I am facing the same problem. Have you been able to solve the problem by now?

DJHansford, 2022-09-08:
In case anyone else stumbles on this, I had the same problem with my Sanodesk controller (HS13A-1), though I've never connected it to an Arduino board. It got stuck on child lock mode and pressing and holding M for 5 seconds didn't work.

I fixed it by pressing and holding the M button while removing the power socket. It beeped at me through this procedure until the residual power in the system died. Just before shutting off it flashed '5-'. Once powered back on it was unlocked. Could just be dumb luck but maybe it'll help someone else.

DJHansford, 2024-08-09:
I've just a variation of this bug, googled for the fix I found before and found my own comment (thanks previous me!). However, this time it was different, my keypad was stuck in locked mode but when I pressed the `M` button it only beeped once, not continuously. The fix above wouldn't work as a result. Instead, after playing around for a long time, I pressed the `1` and `2` buttons together for a few seconds and it beeped continuously at me, then pressing the `M` caused the same beeping and my fix above worked.

fundix, 2024-12-12:
I fixed it by holding M and power off ... still holding M and power on the table ... on display was 888, then I press and hold arrow down and M and table start working