iMicknl/LoctekMotion_IoT #12: Miswired, still works

sharkzor, 2021-09-19:
Not really an issue, but i accidentally wired my CB38M2B(IB)-1 / HS13A-1

with the other wiring scheme.
Still worked! How can this be? According to the scheme it should have zapped the board with 23v and rx/tx doesnt line up.
Does the control box switch or has some handshake protocol ?
[image]

iMicknl, 2021-09-19:
This is interesting! We had a discussion on the Discord a while ago about this, and we were wondering how multiple control panels could have different wirings. Perhaps the control box is able to recognize and switch the signal.

iMicknl, 2021-12-16:
@sharkzor could you perhaps share your wiring? It is hard to define that from the cable color. 

Perhaps you can use the numbers from this image:
[image]

mad2xlc, 2021-12-18:
I have the same controller and control panel CB38M2L(IB)-1 and HS13A-1.
I measure 5V between pin 8 and 7, and between 8 and 4 (what is Pin20 i think).
So it makes no sense for me to use the pinout from https://github.com/iMicknl/LoctekMotion_IoT/blob/main/README.md#hs13a-1

MEfinger, 2021-12-28:
Hey @iMicknl,
I have done the same mistake as @sharkzor and wired my desk (CB38M2L(IB)-1 and HS13B-1) with the [HS01B-1 scheme](https://github.com/iMicknl/LoctekMotion_IoT#hs01b-1). So in my case the following pins are connected:

RJ45 pin | Name | Ethernet cable color (T568B) | Wemos D1
-- | -- | -- | --
8 | +5V (VDD) | Brown | 5V
7 | GND | White-Brown | GND
6 | TX | Green | D5
5 | RX | White-Blue | D6
4 | PIN 20 | Blue | D2

Everthing works as expected. The Wemos D1 gets its 5V from the Brown & White-Brown cable. The Desk gets the commands, so i assume the Rx and TX is correct too. The only problem is that the Desk Height Sensor sometimes changes its state to "Unknown" and the desk doesn't accept commands. So I assume the Pin 20 wiring is wrong, right?

sharkzor, 2021-12-29:
> @sharkzor could you perhaps share your wiring? It is hard to define that from the cable color.
> 
> Perhaps you can use the numbers from this image: [image]

Its the default wiring scheme B

[image]

mad2xlc, 2021-12-29:
In my setup the 5v drops to 4.8v and the wemos is getting unstable. I think you have similar problems. i will try it with an barebone esp8266 and a 3.3v ldo voltage regulator. Otherwise i will need an external power supply.

MEfinger, 2021-12-30:
Out of curiosity I tested the other wiring scheme (HS13A-1) and it did not work. The Wemos wouldn't turn on and the physical Keypad was stuck (The display turned on but I couldn't press any keys). I also tested the other RJ-45 Port to see if that would change something and even disconnected the physical keypad. But the result stayed the same. The moment I plug in a Wemos D1 with the HS13A-1 wiring scheme the Desk gets stuck and wont respond to the physical or the digital keypad. Even powering off the whole desk and then reconnecting the Wemos to the Port wouldn't help. 
Do you guys have any idea on why only the other scheme works?

MEfinger, 2021-12-30:
Hey @iMicknl, 
I think I found the issue with the wrong wiring scheme. 
In the Readme the Pin Layout for the HS13A-1 is wrong.
I opened up my Control Box and compared the values on the PCB and saw that there are actually 10 cables connected to it.
So the Table in the Readme File should be changed (also the ethernet cable colors in the HS13A-1 table are swaped)

RJ45 pin | Name | Original Cable Color | Ethernet cable color (T568B)
-- | -- | -- | --
8 | RESET | Brown | White-Orange
7 | SWIM  | White | Orange
6 | empty | Purple | White-Green
5 | pin20| Red | Blue
4 | RX | Green | White-Blue
3 | TX | Black | Green
2 | GND1 | Blue | White-Brown
1 | +5V (VDD) | Yellow | Brown

The two cable underneath the 5V are separated from the RJ45 cable and belong to the powerplug. they deliver the 29V which I assume are used to power the USB port. Thats why only the HS13B-1 has them.

[image]

iMicknl, 2022-01-24:
@MEfinger I think you are correct! Would you be willing to do a PR to fix this? Or should I do this :).

Which YAML are you using? Do you set PIN20 to low or high?

Kienz, 2022-02-23:
Does anybody got CB38M2L(IB)-1  with HS13U and ESP8266 working? @MEfinger @mad2xlc

dasmith83, 2022-09-07:
I too have a CB38M2B(IB)-1 but my buttons are on a HS11A-1.

I tested the pins on both combinations (i.e. for the HS01B-1 and for the HS13A-1), I also found that for my HS11A-1 the correct configuration was the same as the HS01B-1.

I've used a NodeMCU ESP8266 with the config below:
RJ45 pin 8 as +5V going to Pin "VIN"
RJ45 pin 7 as GND going to Pin "GND"
RJ45 pin 6 as TX going to Pin "D5"
RJ45 pin 5 as RX going to Pin "D6"
RJ45 pin 4 going to Pin "D2"

Thanks to all those who've made this possible...works a treat!

irqnet, 2022-09-15:
> I too have a CB38M2B(IB)-1 but my buttons are on a HS11A-1.
>
> I tested the pins on both combinations (i.e. for the HS01B-1 and for the HS13A-1), I also found that for my HS11A-1 the correct [...]

I do have HS11A-1 connected to a  CB38M2**D**(**P**B)-1 

Configuration seems not to work for me. As there is no dedicated RJ45 Port, I've just wired the Wemos D1 mini to the connector. I've just unsoldered D5 for tests, as the pysical buttons are not responding when everything is connected.

[image]

sickill, 2023-01-21:
I have **CB38M2L(IB)-9** with **HS13B-1** panel. The correct wiring for this combination is the same as for **HS01B-1**:

ESP32 pin    | RJ45 pin    | RJ45 cable color                
-------------|-------------|-------------
+5V          | 8           | brown 
GND          | 7           | white-brown
17 (TX)      | 6           | green
16 (RX)      | 5           | white-blue
23 ("Pin20") | 4           | blue 

Initially I used NodeMCU v3 (ESP8266) for this but I was experiencing some issues:

- it was unstable, resetting frequently - maybe related to the voltage drop to 4.8 @mad2xlc brought up (I measured and I'm seeing similar value)
- height readings had occasional glitches, showing 600.0 cm out of the blue

ESP8266 doesn't have hardware UART so it does it in software and it's not very robust (esphome docs on UART warn about that).

So now I'm using ESP32 DevkitC-32E for this, powering it from pin 8 +5V without any problems. It works flawlessly - always accurate height readings and super stable, never resets. ESP32 DevkitC-32E has LDO voltage regulator which to my knowledge converts it to 3.3V. Maybe that explains why it's stable with 4.8V or even 4.75 I measured on few occasions.

naresh97, 2023-02-03:
> Hey @iMicknl, I think I found the issue with the wrong wiring scheme. In the Readme the Pin Layout for the HS13A-1 is wrong. I opened up my Control Box and compared the values on the PCB and saw that there are actually 10 cables connected to it. So the Table in the Readme File should be changed (also the ethernet cable colors in the HS13A-1 table are swaped)
> RJ45 pin 	Name 	Original Cable Color 	Ethernet cable color (T568B)
> 1 	RESET 	Brown 	White-Orange
> 2 	SWIM 	White 	Orange
> 3 	empty 	Purple 	White-Green
> 4 	pin20 	Red 	Blue
> 5 	RX 	Green 	White-Blue
> 6 	TX 	Black 	Green
> 7 	GND1 	Blue 	White-Brown
> 8 	+5V (VDD) 	Yellow 	Brown
> 
> The two cable underneath the 5V are separated from the RJ45 cable and belong to the powerplug. they deliver the 29V which I assume are used to power the USB port. Thats why only the HS13B-1 has them.
> 
> [image]

Can confirm, works on my HS13B-1.

closed 2023-02-04 by iMicknl

260885El, 2024-03-19:
Good afternoon, query, is there another controller that can replace the model CB38M2B (IB) - 1

Loic691, 2024-06-11:
Hi
Too follow
I rece,tly bought a E7 pro and the hardware is a CB38M2L(IB)-2
Is this the same pinout ?
[image]

iMicknl, 2024-06-11:
@Loic691 it is easier to see if you open your control panel and show the PCB. You can join our Discord to discuss, but good to first do some efforts yourselves.

Loic691, 2024-06-11:
> @Loic691 it is easier to see if you open your control panel and show the PCB. You can join our Discord to discuss, but good to first do some efforts yourselves.

Well. I try to take photo of PCB this week end !

PR #46 (2023-02-04): Includes pinout for HS13B-1
Uses pinout from @MEFinger, fixes #12 

Removed some deadlinks from loctekmotion.com. I couldn't find the corresponding fixed links on the website. They must have removed it.