iMicknl/LoctekMotion_IoT #13: Would this work using an USB to RJ45 adapter using a normal linux machine?

edoardo849, 2021-09-29:
Hello - I was wondering whether an Ethernet / RJ 45 adapter such as [this one](https://www.amazon.co.uk/Network-Adapter-Syncwire-Gigabit-Ethernet/dp/B074CDL78M/ref=asc_df_B074CDL78M/?tag=googshopuk-21&linkCode=df0&hvadid=310847932305&hvpos=&hvnetw=g&hvrand=5586666190401316124&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=1006912&hvtargid=pla-471470781519&psc=1) attached to a linux machine could potentially work. Since the PI uses Linux anyways, the question to me is:

- can we send GPIO commands over the serial port? 
- is there anything "special" or incompatible about a USB to Ethernet adapter?

I might give it a go during the weekend unless someone thinks that there is something fundamentally wrong with this approach.

iMicknl, 2021-09-29:
@edoardo849 I have no experience on this field, however I would doubt that it works. Not sure if your Linux machine can see your RJ45 adapter as a serial input.

iMicknl, 2021-12-07:
Let's close this issue for now, feel free to update us @edoardo849 if you already tried it.

closed 2021-12-07 by iMicknl