iMicknl/LoctekMotion_IoT #15: No Bug or issue - but i think i found the Bluetooth Adapter

NemesorZahndrekhh, 2021-12-01:
Hello, first at all - sorry that i need to use the Bug / Issue section, but i didnt found another way to write this publicly^^ i am new to github :). 

As i am not that into programming i searched for the Bluetooth Adatper and i think i found him. It's from Linak, the same company which produces the controller boxes called CBD6S - every flexispot or other standing desk with these type of control boxes should be compatible to this Bluetooth Adatper - this Adapter gives you access through smartphone app and a windows desktop app (should be in windows store).

The thing is, Linak didn't sell them on their own - or seems to be they just sell their products to other Stores. So i just found a Store and i gonna order one now to try it on my just "new" Flexispot E6.

Here are the links to the Bluetooth adapter and the compatible control box

Bluetooth Adatper: https://www.linak-us.com/products/accessories/bluetooth-adapter-desk/

Control Box: https://www.linak-us.com/products/control-boxes/cbd6s

After i tested this adapter i'll get back to you! :)

iMicknl, 2021-12-07:
Cool @NemesorZahndrekhh, keep us updated! LoctekMotion changed their website and they removed all references to the Bluetooth controller (https://www.loctekmotion.com/shop/accessories/bt-desk-app/)...

I wonder if the Linak one is a different brand, or compatible.

NemesorZahndrekhh, 2021-12-08:
So, today i received the adapter and i got bad news. If i plug it in, it doesn't start to blink like in the Linak tutorial video.

I = cable / O = no cable  (all cable colored white)

8  7  6   5   4  3  2  1
I  O  O  O  O  I   I   I

So i can't really tell the function of the pins

as 4 pins are not in use, do you think it is able to (or better question, it makes more sense to "fix" it and make it useable for this kind of controller - but i need to "destroy" it if i open the case - or better return it)

i upload some images too

(if there's a option which i just need to solder some wires away or add the missing ones, my brother could help me with that)

sure i understand that long range guidance / advice of such things could be tricky and if you try to help me, and i fail because the option doesnt give me the solution i want, i never would blame you :)

images are following:

best regards zahndrekh

iMicknl, 2022-06-27:
Thanks for your detailed report, @NemesorZahndrekhh!

closed 2022-06-27 by iMicknl