iMicknl/LoctekMotion_IoT #19: CB28M1B-1 - up/down switches not working

marcelfischer, 2021-12-28:
Hi all, my controller box is a CB28M1B-1 with a HS01B-1 keypad.
I disconnected the keypad and connected an ESP8266 as described with esphome.
Preset 1-3 and wakeup is working fine. But with up and down nothing happens.
Is there any way to debug this?

Many thanks for this awesome work!

marcelfischer, 2021-12-28:
Ah sorry, those up and down switches were set to internal. I remove that, now its working

iMicknl, 2021-12-28:
You should see a cover with up/down buttons, hence why you don't see the switches. On Discord I shared a newer config with button support as well, however I didn't had time yet to add it to the repo.

marcelfischer, 2021-12-28:
Im not really into Discord. But I will check it. Or are you able to post it here?

iMicknl, 2021-12-29:
https://gist.github.com/iMicknl/2890d738ede31947b81fb33b8951c242

larc-firefly, 2022-12-12:
I have a standing desk with CB28M1S-1 controller, the up button stopped working. It required a reset. The reset was simple, press the lowering button until the desk hits rock bottom, hold it until the motor stops then try’s to raise slightly automatically. The reset is now complete and the up button should work again.

justspacedog, 2024-06-22:
@marcelfischer Could you please provide the pinout for your setup? :)

xdasx, 2024-11-13:
@larc-firefly are you using CB28M1S-1 with esp? 
memory functions working also?
could you share your config?

findCarolinaCosta, 2026-02-08:
> I have a standing desk with CB28M1S-1 controller, the up button stopped working. It required a reset. The reset was simple, press the lowering button until the desk hits rock bottom, hold it until [...]

It worked here, thank you very much.