iMicknl/LoctekMotion_IoT #23: ESP32 D1 mini, CB38M2J(IB)-1 combination

hofvanta, 2022-01-24:
Hi, all,

I have the CB38M2J(IB)-1, with two keypad RJ45 ports. I stripped standard Ethernet cable and used the following scheme:
ESP32	Name	HS13A-1	T568B
GND		GND		Green	White-blue
GPIO16	RX		Purple  White-Green
GPIO17	TX		Red		Blue
GPIO23	PIN20	White	Orange
VIN	+5V (VDD) 	Black	Green

The following issues I have. The VCC from CB38M2J(IB)-1, seems not +5V, but used at the beginning external power. And the ESP32 does not see any activity.

Esphome config used for ESP32 D1 mini:
`substitutions:
  device_name: desk-guestroom
  name: desk-guestroom
  min_height: "73.6" # Min height + 0.1
  max_height: "122.9" # Max height - 0.1

esphome:
  name: ${name}
  comment: ${device_name}
  platform: ESP32
  board: mhetesp32minikit
  includes:
    - desk_height_sensor.h

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${device_name} Fallback Hotspot"
    password: !secret ap_fallback_password

captive_portal:

# Enable logging
logger:
  #level: DEBUG
  baud_rate: 0

# Enable Home Assistant API
api:
  password: !secret api_password

ota:
  password: !secret ota_password

uart:
  id: desk_uart
  baud_rate: 9600
  tx_pin: GPIO17
  rx_pin: GPIO16

sensor:
  - platform: wifi_signal
    name: "WiFi Signal"
    update_interval: 60s

  - platform: uptime
    name: Uptime

  - platform: custom
    lambda: |-
      auto desk_height_sensor = new DeskHeightSensor(id(desk_uart));
      App.register_component(desk_height_sensor);
      return {desk_height_sensor};
    sensors:
      id: "desk_height"
      name: Desk Height
      unit_of_measurement: cm
      accuracy_decimals: 1
      icon: "mdi:counter"

switch:
  - platform: gpio
    name: "Virtual Screen"
    pin:
      number: GPIO23
      mode: OUTPUT
    restore_mode: ALWAYS_OFF
    internal: true

  - platform: uart
    name: "Preset 1"
    id: switch_preset1
    icon: mdi:numeric-1-box
    data: [0x9b, 0x06, 0x02, 0x04, 0x00, 0xac, 0xa3, 0x9d]
    uart_id: desk_uart

  - platform: uart
    name: "Preset 2"
    id: switch_preset2
    icon: mdi:numeric-2-box
    data: [0x9b, 0x06, 0x02, 0x08, 0x00, 0xac, 0xa6, 0x9d]
    uart_id: desk_uart

  - platform: uart
    name: "Sit" # Preset 3 on some control panels
    id: switch_sit
    icon: mdi:chair-rolling
    data: [0x9b, 0x06, 0x02, 0x00, 0x01, 0xac, 0x60, 0x9d]
    uart_id: desk_uart

  - platform: uart
    name: "Stand" # Not available for all control panels
    id: switch_stand
    icon: mdi:human-handsup
    data: [0x9b, 0x06, 0x02, 0x10, 0x00, 0xac, 0xac, 0x9d]
    uart_id: desk_uart

  - platform: uart
    name: "Up"
    id: switch_up
    icon: mdi:arrow-up-bold
    data: [0x9b, 0x06, 0x02, 0x01, 0x00, 0xfc, 0xa0, 0x9d]
    uart_id: desk_uart
    internal: true

  - platform: uart
    name: "Down"
    id: switch_down
    icon: mdi:arrow-down-bold
    data: [0x9b, 0x06, 0x02, 0x02, 0x00, 0x0c, 0xa0, 0x9d]
    uart_id: desk_uart
    internal: true

  - platform: uart
    name: "M"
    id: switch_m
    icon: mdi:alpha-m-circle
    data: [0x9b, 0x06, 0x02, 0x20, 0x00, 0xac, 0xb8, 0x9d]
    uart_id: desk_uart

  - platform: uart
    name: "(wake up)" # Not available on all control panels
    id: switch_wake_up
    icon: mdi:gesture-tap-button
    data: [0x9b, 0x06, 0x02, 0x00, 0x00, 0x6c, 0xa1, 0x9d]
    uart_id: desk_uart

cover:
  - platform: template
    # icon: mdi:table-chair
    # icon: mdi-human-male-height-variant
    name: "Desk"
    assumed_state: true

    # Move desk up
    open_action:
      - while:
          condition:
            sensor.in_range:
              id: desk_height
              below: ${max_height}
          then:
          - logger.log: "Executing up command"
          - switch.turn_on: switch_up
          - delay: 10ms
    
    # Move desk down
    close_action:
      - while:
          condition:
            sensor.in_range:
              id: desk_height
              above: ${min_height} 
          then:
          - logger.log: "Executing down command"
          - switch.turn_on: switch_down
          - delay: 10ms
    optimistic: true`

iMicknl, 2022-01-24:
Would be good to add a pictures of your PCB (of the desk keypad). And do you own a multimeter? You should be able to check if your wiring is correct. In the past I have seen mistakes / issues with just using the color of the RJ45 wires.

You could uncomment https://github.com/iMicknl/LoctekMotion_IoT/blob/a0137cb3ea0a13f324fde7bc63f444deea071357/packages/esphome/desk_height_sensor.h#L85 so that you will see all incoming bytes. This will also help to spot if you are using the right TX/RX.

hofvanta, 2022-01-24:
The keypad is H13A-01.
[image]
I will uncomment the line and let you know.

iMicknl, 2022-01-24:
Could you also add a picture of the PCB? (so screw open your desk keypad).

hofvanta, 2022-01-24:
The PCB is cabled as documented H13A-1.
[image]

The uncomment, helped, I got continious incoming bytes, which default 000000 and changing values by hitting the height buttons of the keypad. Home assistant, see the sensor, but value is Unavailable.

hofvanta, 2022-01-24:
Changing height logging:
[image]

iMicknl, 2022-01-24:
> Did you try this one already? [#12 (comment)](https://github.com/iMicknl/LoctekMotion_IoT/issues/12#issuecomment-1003183711) I think this could be the correct wiring.

??

hofvanta, 2022-01-24:
OK, I will try other wiring as #12 commented.

hofvanta, 2022-01-24:
Seems now to work with other cable color scheme. Tx. The home assistant integration only show the height as soon you hit the keypad and will not update the height after (after unpressing the button). Is there any fix for that?

iMicknl, 2022-01-24:
Great to hear! We need to update the docs soon, great to have this confirmed.

Can you share the YAML you use now? And does the height update when you execute a command (like sit/stand) from your ESPHome? Current behavior is that the height only updates when something has been executed.

That is why in my other example, we call the 'M' button quickly on startup.

hofvanta, 2022-01-24:
Im updating the YAML file now. Because I have more esphome integrations I will change the sensor and switches with desk_<> to seperate them more easily in home assistant. Will send you as soon everything is functioning. Tx for now.

hofvanta, 2022-01-24:
My personal dongle. The nonused wires are protected (thin shrink tubes) and all fixed by using one big heat shrink tube.
[image]
Thanks for the help

hofvanta, 2022-01-24:
The esp integration is working:when use keypad the initial height is recorded to HA. When also the up/down is working. But as soon as the keypad is in sleeping mode, no commands from HA/ESP are working. Should the wake up call should trigger to light on the key pad?

Sander51, 2022-02-15:
I've ordered the flexispot E8 and E7, both have the CB38M2J (IB)-1 control box as well. I've used the https://github.com/iMicknl/LoctekMotion_IoT/issues/12#issuecomment-1003183711 wiring (without connecting RESET, SWIM , empty) and all functions work as @hofvanta mentioned as well. The only problem I have as well is once the display goes off it is not possible to control the desk anymore. The wakup call doesn't work. I've also tried to only connect the esp32 and disconnect the remote, but it still doesn't work. @hofvanta did you find a way to solve the sleeping mode issue?

Kienz, 2022-02-21:
> I've ordered the flexispot E8 and E7, both have the CB38M2J (IB)-1 control box as well. I've used the [#12 (comment)](https://github.com/iMicknl/LoctekMotion_IoT/issues/12#issuecomment-1003183711) wiring (without connecting RESET, SWIM , empty) and all functions work as @hofvanta mentioned as well. The only problem I have as well is once the display goes off it is not possible to control the desk anymore. The wakup call doesn't work. I've also tried to only connect the esp32 and disconnect the remote, but it still doesn't work. @hofvanta did you find a way to solve the sleeping mode issue?

How did you connect your ESP32? RJ45 Splitter into HS Port? Because the second RJ45 Port doesn’t work. Can you confirm?

hofvanta, 2022-02-21:
Stefan,

I didnt use any splitter. I used the second port.
I use PIN 16&17 on a Wemos D1 mini esp32, board: mhetesp32minikit

uart:
  id: desk_uart
  baud_rate: 9600
  tx_pin: GPIO17
  rx_pin: GPIO16

switch:
  - platform: gpio
    name: "Virtual Screen"
    pin:
      number: GPIO23
      mode: OUTPUT
    restore_mode: ALWAYS_ON
    internal: true

I hope that it helps to get it work.

Arjan

Kienz, 2022-02-21:
@hofvanta Thanks. Did your control panel work on the 2nd serial port? Mine only works on the first (upper) port. If it’s connect to 2nd port pressing a button only lights up the display but nothing happens. All buttons the same. Did someone have this issue too?

Sander51, 2022-02-22:
@Kienz For the Flexispot E7 and E8 both control ports work. I did have some issues though with the control panel while fixing this and cutting off power to the desk for a while and reconnecting fixed it, never had the issue again.

@hofvanta Thanks. For me it turns out a slight edit of the flexispot_e5b_esp32.yaml code seems to work perfectly for the E7 and E8. I'm waiting for an utp splitter so I can also connect the control panel directly so I would know the current state of the desk if only the control panel is used.

hofvanta, 2022-02-22:
No, both ports are working without any issues.

Kienz, 2022-02-22:
@hofvanta Strange. I contacted the Flexipot support. Let's see what they say.
Other question: Should it work if I only connect the eg ESP8266 to first port? (Display is disconnected) Can someone test this. Kind regards.

oischinger, 2022-12-04:
@Sander51  did the UTP splitter work out as expected? Does the ESP32 now always output the correct position even when the desk was operated via the panel?