iMicknl/LoctekMotion_IoT #24: ASR

DeastinY, 2022-01-26:
I've been using the ESPHome variant for a few days and today set up an automation to press M1 / M2 every 30 minutes. After some time it failed and the display showed "ASR".  
Couldn't control anything with the ESP anymore and needed to attach the panel and move the desk all the way down to reset it.
Anyone already had similar issues? Something to expect to happen more often?

DeastinY, 2022-06-28:
I've encountered the issue so often that I stopped using it altogether :/  
In the short time I tried to look into it I couldn't really get any information together, so I just stopped.
Willing to help if anyone wants to look into this though, if time permits.

xkisu, 2022-07-11:
I'm getting this frequently as well too, I've tried adjusting the esp home config a bit to cover what I thought where some edge cases with the cover but it hasn't helped.

Desk: PrimeCables PLUS+
Control Panel: Loctek HS01B-1

It seems to get stuck in the ASR/Reset mode any time I leave it and don't move the desk for several hours. 

On a side note, I added a button component to reset it (but I've had intermittent success with wether it works or not, sometimes you hear the relay clicking to move the desk down but it doesn't actually move):
```yaml
button:
  ...
  - platform: template
    name: "Reset Desk"
    on_press:
      - while:
          condition:
            sensor.in_range:
              id: desk_height
              # When the desk is in ASR/Reset mode is appears to return an invalid height of 50cm.
              #
              # When this happens, the down button needs to be held until the desk lowers all the way
              # and the height resets to the correct minimum height.
              below: ${min_height}
          then:
            - logger.log: "Executing down command"
            - button.press: button_down
            - delay: 10ms
```

xkisu, 2022-07-11:
I noticed that my ESP was periodically rebooting it's self as I'm not using Home Assistant (using my own API client) and the [Native API Component](https://esphome.io/components/api.html) is setup to reboot the entire board after 15mins by default if a Home Assistant client hasn't connected.

I've disabled the automatic reboot to see if that improves the issue, I'm wondering if my modified ESP config pressing M button on boot every 15 minutes due to the restart is what's been causing it to happen frequently.

xkisu, 2022-07-12:
I made a few changes to the esphome config that seem to have fixed the ASR code repeatedly appearing, I've left it running for 48 hours and made constant height adjustments and haven't ran into it again, whereas previously it would happen every couple of hours.

Changes I made:
 * Fixed the board rebooting every 15 minutes due to no Home Assistant connection
 * Changed the UART switch components out for buttons that send a UART command, not sure if this was an improvement but it simplified some of my other automations
 * Raised the delay on opening/closing the cover to 250ms in case the controller was being overwhelmed, I noticed some odd behaviour when it was lower (like the desk "seizing" up, shaking, and not really moving) that's also stopped now that it's higher

One other thing I'm also going to add if the ASR code occurs again is toggling the "Virtual Screen" pin high/low on a timeout when buttons are pressed to simulate the actual behaviour I'm seeing on the controller in case the screen being alway on to the controller is problematic.

xkisu, 2022-07-18:
My ESPhome has been running for 5 days straight now with no issues and I haven't ran into the ASR code again, I think increasing the button delay on the cover to 250ms and fixing the board rebooting every 15 minutes definitely solved whatever was causing it to occur constantly.

DeastinY, 2022-08-22:
@xkisu did you publish your changes somewhere or could you point me to the changes you did in detail?
Cheers and thanks for apparently figuring out what was going on!

xkisu, 2022-08-22:
> @xkisu did you publish your changes somewhere or could you point me to the changes you did in detail? Cheers and thanks for apparently figuring out what was going on!
> 
> What I figured so far:
> 
> ```
> api:
>   [...]
>   reboot_timeout: 0
> ```
> 
> and
> 
> ```
> cover:
>   - platform: template
>     [...]
>     open_action: #same for close_action
>       - while:
>        [...]
>        -delay: 250ms
> ```

@DeastinY You've got it! I forgot to push, but those are the two changes I made that seem to have stabilized it long-term. I've been running it without issues over a month now after doing those two changes.

DeastinY, 2022-08-23:
[image]
While I can still see strange uptime behaviour in HA, it seems to work stable for me now too! Closing for now, thanks again @xkisu !

xkisu, 2022-08-23:
> [image]
> While I can still see strange uptime behaviour in HA, it seems to work stable for me now too! Closing for now, thanks again @xkisu !

I'm getting that as well on my ESP8266! I'm not entirely sure why, but it seems to be something related to ESPHome and now anything to do with the desk script as far as I can tell 😀

DeastinY, 2022-08-26:
Happened again. Two days it was stable, now height is locked at 50cm and I need to reattach the control panel to change anything :/

KrX3D, 2022-09-25:
Hello, it seems "you" are pressing the M button to long. if it is pressed for 5 seconds (could also be 3 sec depending on your desk) your table enters Factory reset mode, id you go into this mode you need to move you table all the way down to finish resetting.

you could look at my config here. and if your tables does nothing with the 1+2 press comand you could change your code and use this switch and not the M button

https://github.com/iMicknl/LoctekMotion_IoT/issues/35

DeastinY, 2022-09-26:
Hi,
> "you"
why is this quoted 🤨?

> if it is pressed for 5 seconds (could also be 3 sec depending on your desk) your table enters Factory reset mode, id you go into this mode you need to move you table all the way down to finish resetting.

Explanation sounds good, but I'm not pressing the M button manually :/ Is this done automatically somewhere?

Cheers and thanks for the input!

DeastinY, 2022-09-27:
> hi, with "you" i meant some code, since you didnt post your full code i dont know what you changed.
> 
> so maybe your code for pressing M1/M2 every 30 minites is the cause?

aaah, got it, thanks!
Hmm... 
I'm just hitting the M1/M2 via automation same as I would in the UI, my code is unchanged from the original yaml:
```
  - platform: uart
    name: "Preset 1"
    id: switch_preset1
    icon: mdi:numeric-1-box
    data: [0x9b, 0x06, 0x02, 0x04, 0x00, 0xac, 0xa3, 0x9d]
    uart_id: desk_uart
```
I'll dig around, but not really sure it could be this