iMicknl/LoctekMotion_IoT #25: Add support for HCB103A-1

Zeustopher, 2022-01-26:
I love the idea of this project and would love to have support added for the model HCB103A-1 controller. It appears that this is one of the newer models that they are using in several of their desks. What can I do to help get what is needed to add support for the HCB103A-1?

-------------

Example desks:
* https://www.amazon.com/Flexispot-Workstation-Ergonomic-Controller-Adjustable/dp/B07W8W8XKV/
* https://www.amazon.com/FLEXISPOT-Computer-Electric-Controller-Adjustable/dp/B08THG96PV/
* https://www.loctekmotion.com/Standing-Desks/Home-Solutions/ET119
* https://www.flexispot.com/height-adjustable-desks/seiffen-spliced-l-shaped-standing-desk-ef1l

I wasn't able to find the model listed on Flexispot's or LoctekMotion's websites.
An example desk with documentation for the HCB103A-1: https://www.flexispot.my/wp-content/uploads/2021/08/EF1-Manual.pdf

-------------

## Images

[image]

[image]

[image]

[image]
(The port shown to the right labeled as "HS" is the RJ45 port)

Floyer007, 2022-03-14:
I've a similar desk and want to see it in this project as well.

The control panel optically looks exactly the same but is HCB103A-2.
Board has the silkscreen: HCB103(HCB123)_V7(main)

What I figured out so far.
MCU is STM8L151 C8T6

**RJ45**
- Pin 7 - GND

**Four Pin on side**
1. STM Pin 1 - PA0 - SWIM - USART1
2. STM Pin 2 - PA1 - NRST
3. GND
4. VCC (STM Pin 10,11,12)

Floyer007, 2022-03-15:
Well, I created a adapter to breakout the RJ45 to 2.54mm THT Headers and checked the signals with my oscilloscope.
I haven't seen usefull signals on the pins. All tested while moving the desk up and down.

1. to 4. 50mV 50Hz signal -> Noise
5. Steady 5.48V rail
6. Steady 2.7V rail
7. GND
8. Steady 5.74V rail

I will check the four-pin port in the next days.

Zeustopher, 2022-03-15:
@Floyer007 thanks for the help on this. I hope we can come up with something. I had kicked around the idea of just soldering some wires onto the push buttons and doing it that was also, but its just nowhere near as elegant.

Floyer007, 2022-03-15:
Would at least be easier/cleaner on yours, as they use a FPC-cable in my version now:

[image]
[image]

Floyer007, 2022-03-15:
After checking again [HS01B-1](https://github.com/iMicknl/LoctekMotion_IoT#hs01b-1) could fit.
GND [7] is at the right place and 5V [8] (5.7V actally from a internal 3.5A DC-DC converter) as well.

Pin 6 can be routet over D19 and Q5 to STM-Pin 41 (PC2, actually RX).
I wasn't able to trace the other signals on the board.

Setting Pin 4 to 5V didn't initiated a data output nor turned the display on.
Have to try the mentioned [WakeUp Command](https://github.com/iMicknl/LoctekMotion_IoT#retrieve-current-height)

Floyer007, 2022-03-16:
Sadly there is no output on the potential RX and TX pins. (Tried with two serial adapters at my PC via hterm)

- WakeUp command (0x9b 06 02 00 00 6c a1 9d) has no effect
- Reversed coommand (0x9d a1 6c 00 00 02 06 9b) has no effect
- RX, TX switched has no effect

Currently I assume the port is deactivated completely.

Also there is no output on Pin 1 and 2 of the side port.

schmic, 2022-10-19:
I have the same thing as @Floyer007 and found out that the RJ45 was supposed to carry a Bluetooth dongle.

https://fccid.io/2ARK8-BTD02

I tried to fiddle around with an ESP8266 but could not get anything to work, maybe one of you guys can figure out some more information about the BT dongle.

schmic, 2022-10-19:
https://fccid.io/2ARK8-BTD02/Internal-Photos/Internal-Photos-4694579

This should be the interesting part cause it shows the cables for 5V, G, TX & RX.

Floyer007, 2022-10-20:
The Pinout of this dongle just proves [HS01B-1](https://github.com/iMicknl/LoctekMotion_IoT#hs01b-1).
It shows a permanent connection between Pin8 and Pin4.

**Bluteooth-Dongle Pinout**
1 -
2 -
3 (wh) SWS
4 (rd) 5V-SW -> Permanent to +5V
5 (gr) RX-1
6 (sw) TX-1
7 (bl) GND
8 (ye) +5V

Now this dongles protocol would be interesting.
Do you know if it can be purchased somewhere?

schmic, 2022-10-20:
All I found was this video: https://www.facebook.com/Flexispot-FR-103607731866634/videos/cool-avec-un-bluetooth-vous-pouvez-r%C3%A9gler-lhauteur-de-votre-bueau-flexispot-sur-/366861188944567/

No place to buy one of those though (yet).

Floyer007, 2024-03-19:
Hi guys,

I just had this project in mind again and tried the new ESPHome method, but it did not work as well.

So I gave up using the serial interface and did it the classical way.

I utilized the free space, where no components are placed for a second motor.
Everything got isolated with Koptan tape and a Raspberry Pi Pico W placed onto it.
As I learned the hard way, the plastic housing might press the BOOTSEL button. So keep the Pico away of the side of the housing.
The Pico W was choosen, as it's compatible with ESPHome and has a lot of IO, which might be usefull in the future to decode the display as well.

The original plan was to just connect the buttons to GPIOs and use the Picos open-drain to pull them down and press the buttons.
Turns out either the Pico does not support open-drain or ESPHome has issues setting it...

Therefore I choose nFETs (BS170) from an old project to pull the buttons to GND.
Source connected to GND, Gate to a GPIO and Drain via a wire to the resistors next to the buttons.
It does not look nice, but does it's job.
The signals should also be available on R1 and R2 on the main PCB, too.

[image]
[image]

[image]
[image]
[image]

Here is the used ESPHome code I used.
I utilizes the secret functionality:
```
esphome:
  name: flexispot
  friendly_name: flexispot

rp2040:
  board: rpipicow
  framework:
    # Required until https://github.com/platformio/platform-raspberrypi/pull/36 is merged
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

############################
# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_encryption_key

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Flexispot Fallback Hotspot"
    password: !secret fallback_password

time:
  - platform: sntp
    id: sntp_time
    servers:
      - !secret ip_sntp

#captive_portal:

# Not supported on Pico W
#web_server:
#  port: 80

############################
switch:
  - platform: gpio
    id: button_up
    name: "Up"
    icon: "mdi:arrow-up-bold-box"
    pin:
      number: 21
      inverted: false
      mode: OUTPUT
  
  - platform: gpio
    id: button_down
    name: "Down"
    icon: "mdi:arrow-down-bold-box"
    pin:
      number: 20
      inverted: false
      mode: OUTPUT

  - platform: gpio
    id: button_m1
    name: "M1"
    icon: "mdi:numeric-1-box"
    pin:
      number: 19
      inverted: false
      mode: OUTPUT
    on_turn_on:
    - delay: 500ms
    - switch.turn_off: button_m1

  - platform: gpio
    id: button_m2
    name: "M2"
    icon: "mdi:numeric-2-box"
    pin:
      number: 18
      inverted: false
      mode: OUTPUT
    on_turn_on:
    - delay: 500ms
    - switch.turn_off: button_m2

  - platform: gpio
    id: button_m3
    name: "M3"
    icon: "mdi:numeric-3-box"
    pin:
      number: 17
      inverted: false
      mode: OUTPUT
    on_turn_on:
    - delay: 500ms
    - switch.turn_off: button_m3

  - platform: gpio
    id: button_m4
    name: "M4"
    icon: "mdi:numeric-4-box"
    pin:
      number: 16
      inverted: false
      mode: OUTPUT
    on_turn_on:
    - delay: 500ms
    - switch.turn_off: button_m4

```

I hope this gives inspiration and/or helps.

ArthFink, 2024-04-28:
I have tried @Floyer007 approach with an esp d1 Mini, and got it to work without nFETs, not all pins on the esp work, or the connection that I soldered is bad. But I was fine with only having 2 memory buttons and Up and Down. 

``` yaml
switch:
  - platform: gpio
    name: "Position 1"
    icon: mdi:chair-rolling
    id: button_m1
    pin:
      number: D2
      inverted: True
      mode:
        output: True
        open_drain: True
    on_turn_on:
    - delay: 500ms
    - switch.turn_off: button_m1

  - platform: gpio
    name: "Down"
    icon: "mdi:arrow-down-bold-box"
    id: button_down
    pin:
      number: D3
      inverted: True
      mode:
        output: True
        open_drain: True  

  - platform: gpio
    name: "UP"
    icon: "mdi:arrow-up-bold-box"
    id: button_up
    pin:
      number: D4
      inverted: True
      mode:
        output: True
        open_drain: True

  - platform: gpio
    name: "Possition 4"
    icon: mdi:human-handsup
    id: button_m4
    pin:
      number: D6
      inverted: True
      mode:
        output: True
        open_drain: True
    on_turn_on:
    - delay: 500ms
    - switch.turn_off: button_m4
    
   ```