iMicknl/LoctekMotion_IoT #30: [No issue] working setup for Home Assistant + ESPHome + esp32dev + CB28M1B-1 + HS13A-1

guevara777, 2022-03-22:
**Hardware**
to do...

**Wiring**
[image]

**ESPHome**
to do...

marylein, 2022-08-20:
Hi, in the image you connect GPIO 18 and 19, but in the code you define the tx and rx for 16 and 17. Is that just a typo?
I'm using the exact same devices but I can't get this setup to work...

guevara777, 2022-08-22:
The Image is wrong. In the code it´s right. I use the GPIO-Pins 16 and 17 like this:

```
uart:
  id: desk_uart
  baud_rate: 9600
  tx_pin: GPIO17
  rx_pin: GPIO16
```

Here is the correct wirering:
[image]

I also added a position-action since i posted it here. You can populate the cover to Alexa and use voice commands like "set the desk to 70%". 

for this you have to add this to the yaml-file:

```
    # Move desk to desired position
    position_action:
    #Check if we need to move desk up or down from current position
      if:
        condition:
        #Current height is more than target height, then move desk down
          lambda: |-
            float position = (float(id(desk_height).state) - float(${min_height})) / (float(${max_height}) - float(${min_height}));
            return pos < position;
        then:
          - logger.log: "Schreibtisch muss runter gefahren werden"
          - while:
              condition:
                # Move down till the current height is smaller then the target height
                lambda: |-
                  float position = (float(id(desk_height).state) - float(${min_height})) / (float(${max_height}) - float(${min_height}));
                  return pos < position;
              then:
                - logger.log: "Executing down command"
                - cover.template.publish:
                    id: desk_cover
                    current_operation: CLOSING
                - uart.write:
                    id: desk_uart
                    data: [0x9b, 0x06, 0x02, 0x02, 0x00, 0x0c, 0xa0, 0x9d]
                - delay: 108ms
        else:
          - logger.log: "Schreibtisch muss hoch gefahren werden"
          - while:
              condition:
                # Move up till the current height is bigger then the target height
                lambda: |-
                  float position = (float(id(desk_height).state) - float(${min_height})) / (float(${max_height}) - float(${min_height}));
                  return pos > position;
              then:
                - logger.log: "Executing up command"
                - cover.template.publish:
                    id: desk_cover
                    current_operation: OPENING
                - uart.write:
                    id: desk_uart
                    data: [0x9b, 0x06, 0x02, 0x01, 0x00, 0xfc, 0xa0, 0x9d]
                - delay: 108ms
```

xXkr13g3rXx, 2022-12-22:
Hi guys. If I add the second part (Move desk to desired position) i get a double definition error from position_action.

I‘m a noob in coding and do need help there. Can anyone help me?