iMicknl/LoctekMotion_IoT #32: Help with ESP32 HS01B-1 + CB22M2A-1

copystring, 2022-08-09:
Hello everyone.

So for some days I tried to get this working.
I wired up my hardware with passthrough like this:
[image]
Taken from here: https://github.com/iMicknl/LoctekMotion_IoT/tree/main/packages/esphome#hs01b-1
Correct me if my wiring is wrong. Maybe I misunderstood the wiring for passthrough with esp32.

It does not work. Display stays dark. Can't control it via home assistant. The buttons also don't work.
I checked my wiring a lot of times with multimeter. I'm certain it is correctly wired up just like on the picture.

Sometimes I had luck (2-3 times) after replugging HS01B-1 the display turns on and the table height is published in home assistant.

I'm sure my wiring is correct. It must be something else.

Please help me so I can sleep at night again :) 😅

copystring, 2022-08-14:
So I sort of managed to get this working. I rebuild the setup with two of those: 
[image]

Wired everything up again. It sort of works. I can control the table via MQTT with the esphome yaml file for passthrough.
However the esp32 crashes a lot when sending commands via MQTT. Also passthrough does not work.

tobit6795, 2023-08-12:
@copystring could you give some more information? im fighting wwith cb27m1b-1 + hs01b-1 for hours - i want to sleep at night again, too :) would appreciate any help