iMicknl/LoctekMotion_IoT #34: Help figuring out possible support for CB38M2D(PB)-2

Scoff123, 2022-09-11:
I'm hoping to work out whether or not this can be used with my Flexispot EQ5 model which uses the CB38M2D(PB)-2 control unit with Sanodesk LCD controller. 
The control unit doesn't have a spare RJ45 port. The LCD button panel connects to the Control box with one cable which splits into two - an RJ45 and a 2 pin connector which both go into the control box. 
The units appear to be sealed/glued with no easy access so I can't open them to view the boards inside, and with it still being under warranty I'm a bit hesitant to try and split the cases! 
Pictures to show the units, but I suppose it's a long shot hoping to get this working unless anyone else with the same model has been brave enough to open the units up to investigate? 

[image]
[image]
[image]
[image]
[image]

IamMattM, 2023-05-19:
[image]

Scoff123, 2023-05-19:
Thanks for posting up pictures of your board, and for the information you've provided. I've not managed to make any progress myself, however I will try to open my controller up and check the board model to see if it's any different to yours. Hopefully someone with more knowledge than myself in this area can help to ascertain whether support of this controller will indeed be possible.

IamMattM, 2023-05-19:
I figured out after looking at datasheet for the STM32 part above that *if* there was going to be a second UART port that would accept any control commands, it would have to be UART#2 since UART#2 is in use and attaches to the RJ45 port.

UART#2 RX/TX pads are currently untracked so it should be possible to send a few commands to it and see if there are any responses....

[image]

IamMattM, 2023-05-19:
UART#1 goes to RJ45 (while SWD JTAG debug/progrm goes to edge contacts)
[image]

rsiuda, 2023-05-22:
I’ve received the same control box (and was hoping for 2 ports). 

Did you try to use the ESP32 instead of the LCD Panel? Would that work?

IamMattM, 2023-05-22:
I have not got the opportunity to connect an ESP32 to the one-and-only RJ45 port.

Was first trying to figure out the details for the box...

rsiuda, 2023-05-22:
I'll go ahead and test it once the esp will be delivered. 

The LCD panel works fine without connecting the 2-pin connector (what is that for?). 

I'll try using an RJ45 splitter to connect both the esp and lcd panel to the box.

Scoff123, 2023-05-22:
Great, I look forward to seeing the results. I've also unplugged the 2 pin connector and all functions on the lcd controller work as normal, so I'm also interested in understanding what that connector is used for.

rsiuda, 2023-05-22:
I think it's powering the usb (charging) port

IamMattM, 2023-05-22:
I imagine that with only one controller box port available for the EQ5 , we will need ot use the "passthrough" mode of an ESP32 as per:

[image]

IamMattM, 2023-05-23:
I did a bit more investigations on the pinout for the EQ5 RJ45 port and scoped the lines using a back to back RJ45 adapter between the keypad and the EQ5 control box.

WARNING: The UART lines are toggling at the full +5V IO swing ! 
Note: ESP32 max IO levels are quoted at +3.6Vmax (3.3V+0.3V) => there is the possiblity that the ESP32 IOs (inputs) will eventually die after being exposed to incoming +5V levels.

PIN20 being an always idling +5V line might be the one most at risk of damaging GPIO22 in the table above.

For now I have worked out the EQ5 to ESP32 and will to the second section which is Kpad to ESP32 later on...

[image]

Scoff123, 2023-05-23:
Great work and thanks for all of your investigations and contributions so far! I'm going to open up the the case of my CB38M2D(PB)-2 to see what is different to the CB38M2A-1 - hopefully it won't be too dissimilar. I've got an ESP32 DEVKIT V1 board spare, so hopefully with level shifting and RJ-45 connectors, plus a case like the printable one you linked, this is going to be achievable...

IamMattM, 2023-05-23:
With regards to the need (or not) to protect the ESP32 from the 5V levels on its IO inputs:

https://www.qworqs.com/2021/05/19/are-the-esp32-and-esp8266-5v-tolerant-yes-they-officially-are/

I am happy to take the very small risk and will wire directly as much more convenient.
Besides, there seems to be plenty of people having gone ahead and wired their ESP32 to the Flexispot lines (which I imagine are all 5V levels) and I have not come across someone saying their ESP32 died in the process.

I will go ahead and make the necessary conenctions for testing but it seems from the yaml file that it should just work...

IamMattM, 2023-05-24:
A small update to say that the 3D print pointed in the post above seems great for the task required: just need to wire things out correctly now between the two keystone RJ45 sockets and ESP32.
One port dedicated to EQ5 connection (need extra ethernet cable) and second port reciving the lead from the keypad....

Fingers crossed.

[image]

  
[image]

Scoff123, 2023-05-24:
Wow that looks perfect and could easily pass as part of the factory setup! Great work. 

I've taken the rubber feet off my control box and unscrewed the cover to check out the board inside. It looks a bit different to your board, and strangely has a reference of ET201-PB, rather than the CB38M2D(PB)-2 which is on the plastic housing...

[image]
[image]

I'll have to see what the reference is on the chip with the green paint, and try to find the data sheet for it.....

Scoff123, 2023-05-24:
[image]

IamMattM, 2023-05-24:
> [image]
> 
> and the datasheet for it https://www.mouser.co.uk/datasheet/2/389/stm8s007c8-1851529.pdf

Thanks - VEry intersting that for the same EQ5 model it seems that the controller board uses various flavours of the STMxx microcontroller.

Saying that, with the "pass-through" ESP32 method/wiring,  as long as the UART out of the control bo tals at the same BAud rte of 9600 and accepts the same commands as the other ones, it should very much "just" work.

IamMattM, 2023-05-24:
Looks like the commands to the EQ5 controller out of the ESP32 are working and I can control the motors up and down.

The keypad and the ESP32 both receive the motor height information from EQ5 controller box and will display height on both keypad and homeassistant.

I have not had a great deal of opportunities to test further but at first glance it is for now encouraging..

The control from the keypad into ESP32.UART0_RX do not appear to be bridged through on ESP32.UART2_TX towards the EQ5 control box so for the time being I was not able to instruct the motor to change height from the keypad. 

I need to see if the ESP32 could bridge those keypad messages and redirect towards EQ5.

Anyhow, some progress...

[image]
[image]

Video showing control from Homeassistant with height info on keypad....some quirks to understand/look at:
https://photos.app.goo.gl/NAHssQVEAeAsj6nJ7

Scoff123, 2023-05-25:
Well done, looks like you are nearly there! Great to see it in action, and hopefully you can iron out the issue with the keypad commands not reaching the EQ5 controller box. 

I don't have a scope so probably can't take the testing on my own board much/any further, so I'm just going to go with the same setup that you've implemented and give it a test. I'll grab a couple of keystone jacks online, flash my ESP32 board and add it to HA,  and try connecting things up and see what happens. This has also convinced me it's time to finally invest in a 3D printer :)

IamMattM, 2023-05-25:
Yes, I also did not have a 3D printer until a couple of months ago when there was an offer for the Ender 3 v2 NEO -
Since then I migrated the 3D printer to "klipper+klipper screen android" and have been making full use of it for all those IOT/Smart devices enclosures that I need. Well worth it.

I'll try to figure out where things are not working by comparing the various yaml made available...

I can already see that in my HA integration I do not see the "STOP" button like in the following reference picture:

[image]

IamMattM, 2023-05-25:
An update with regards to verifying the commands sent out by keypad:
https://drive.google.com/drive/folders/17fUi4JQyRUNruGem4oYe32flX-6Btzhs?usp=sharing

When the  keypad when *not* pressed, it is correctly sending out continuously the "Empty" command, 

When pressed,on the various buttons (Up / Down / Preset1=stand / Preset2=sit/ Preset3 /_________/ Memory) it is also correctly sending out the correct command sequence.
*Note: THere is one button that is not yet implemented in the yaml file and that is the PReset4 with command :
[0x9b, 0x06, 0x02, 0x00, 0x01, 0xac, 0x60, 0x9d] as captured below:

[image]

With the doubt as to whether the keypad was actually sending anything when pressed on the line going to ESP32.UART0<RX>, I can now say for certain that this is happening.

What is NOT happening however is any software "forward" activity of those commands to the ESP32.UART2<TX> towards the EQ5...

Maybe the UART bridging is not supported....

@pepsinio @iMicknl : would you be able to comment on the matter of esp32 passthrough for the Loctek Motion keypads using ESPHOME esp32_e5b.yaml ?

I am not having much success....

Scoff123, 2023-05-26:
So the keypad is definitely sending the commands correctly the the ESP board. Are you able to view debug logging (logger.log) or print the debug messages (after changing the baud_rate from zero in the ESPHome config yaml) to determine if the ESP is actually receiving these sent keypad commands?

IamMattM, 2023-05-26:
@Scoff123 : THanks for the pointer and enabling Baud= 9600 (from 0)
I can see the messages when pressing keypad being decoded correctly (except for Preset4 not yet implemented)

`[12:49:39][C][mdns:109]:   Hostname: esp32-flexispot-eq5
[12:49:39][C][ota:093]: Over-The-Air Updates:
[12:49:39][C][ota:094]:   Address: esp32-flexispot-eq5.local:3232
[12:49:39][C][ota:097]:   Using Password.
[12:49:39][C][api:138]: API Server:
[12:49:40][C][api:139]:   Address: esp32-flexispot-eq5.local:6053
[12:49:40][C][api:141]:   Using noise encryption: YES
[12:49:40][C][wifi_signal.sensor:009]: WiFi Signal 'WiFi Signal'
[12:49:40][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[12:49:40][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[12:49:40][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[12:49:40][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[12:49:50][D][sensor:094]: 'Desk command': Sending state 3.00000  with 0 decimals of accuracy
[12:49:50][D][switch:012]: 'Preset 1' Turning ON.
[12:49:50][D][main:243]: Executing Preset 1 command
[12:49:50][D][main:746]: Executing Empty command
[12:49:50][D][main:322]: Executing screen timer command
[12:49:50][D][switch:012]: 'Virtual Screen' Turning ON.
[12:49:50][D][sensor:094]: 'Desk command': Sending state 8.00000  with 0 decimals of accuracy
[12:49:54][D][sensor:094]: 'Desk command': Sending state 4.00000  with 0 decimals of accuracy
[12:49:54][D][switch:012]: 'Preset 2' Turning ON.
[12:49:54][D][main:308]: Executing Preset 2 command
[12:49:54][D][main:746]: Executing Empty command
[12:49:54][D][script:077]: Script 'screen_timer' restarting (mode: restart)
[12:49:55][D][main:322]: Executing screen timer command
[12:49:55][D][switch:012]: 'Virtual Screen' Turning ON.
[12:49:55][D][sensor:094]: 'Desk command': Sending state 8.00000  with 0 decimals of accuracy
[12:49:56][D][sensor:094]: 'Uptime': Sending state 32.44800 s with 0 decimals of accuracy
[12:50:01][D][sensor:094]: 'Desk command': Sending state 5.00000  with 0 decimals of accuracy
[12:50:01][D][switch:012]: 'Preset 3' Turning ON.
[12:50:01][D][main:373]: Executing Preset 3 command
[12:50:01][D][main:746]: Executing Empty command
[12:50:01][D][script:077]: Script 'screen_timer' restarting (mode: restart)
[12:50:02][D][main:322]: Executing screen timer command
[12:50:02][D][switch:012]: 'Virtual Screen' Turning ON.
[12:50:02][D][sensor:094]: 'Desk command': Sending state 8.00000  with 0 decimals of accuracy
[12:50:03][D][sensor:094]: 'WiFi Signal': Sending state -46.00000 dBm with 0 decimals of accuracy
[12:50:09][D][sensor:094]: 'Desk command': Sending state 6.00000  with 0 decimals of accuracy
[12:50:09][D][switch:012]: 'M' Turning ON.
[12:50:09][D][main:438]: Executing Preset 3 command
[12:50:09][D][main:746]: Executing Empty command
[12:50:10][D][script:077]: Script 'screen_timer' restarting (mode: restart)
[12:50:10][D][main:322]: Executing screen timer command
[12:50:10][D][switch:012]: 'Virtual Screen' Turning ON.
[12:50:10][D][sensor:094]: 'Desk command': Sending state 8.00000  with 0 decimals of accuracy`

IamMattM, 2023-05-26:
And it actually worked once for Preset1 and Preset2 (moving feet from keypad)  at some point but is now no longer working although the commands are always being decoded correctly by ESP32.

IamMattM, 2023-05-26:
When presseing the "up" toggle on the HA integration page I get the following messages and the feet move up:

[13:00:47][D][switch:012]: 'Virtual Screen' Turning ON.
[13:00:52][D][switch:012]: 'Up' Turning ON.
[13:00:52][D][switch:055]: 'Up': Sending state ON
[13:00:52][D][uart.switch:020]: 'Up': Sending data...
[13:00:52][D][switch:055]: 'Up': Sending state OFF
[13:00:52][D][sensor:094]: 'Desk Height': Sending state 62.10000 cm with 1 decimals of accuracy
[13:00:53][D][script:077]: Script 'screen_timer' restarting (mode: restart)
[13:00:53][D][main:322]: Executing screen timer command
[13:00:53][D][switch:012]: 'Virtual Screen' Turning ON.
[13:00:53][D][sensor:094]: 'Desk Height': Sending state 62.20000 cm with 1 decimals of accuracy
[13:00:53][D][script:077]: Script 'screen_timer' restarting (mode: restart)
[13:00:53][D][main:322]: Executing screen timer command
[13:00:53][D][switch:012]: 'Virtual Screen' Turning ON.
[13:00:53][D][sensor:094]: 'Desk Height': Sending state 62.30000 cm with 1 decimals of accuracy
[13:00:53][D][script:077]: Script 'screen_timer' restarting (mode: restart)
[13:00:53][D][main:322]: Executing screen timer command
[13:00:53][D][switch:012]: 'Virtual Screen' Turning ON.
[13:00:53][D][sensor:094]: 'Desk Height': Sending state 62.40000 cm with 1 decimals of accuracy
[13:00:53][D][script:077]: Script 'screen_timer' restarting (mode: restart)
[13:00:53][D][main:322]: Executing screen timer command
[13:00:54][D][switch:012]: 'Virtual Screen' Turning ON.
[13:00:54][D][sensor:094]: 'Desk Height': Sending state 62.50000 cm with 1 decimals of accuracy
[13:00:54][D][script:077]: Script 'screen_timer' restarting (mode: restart)
[13:00:54][D][main:322]: Executing screen timer command
[13:00:54][D][switch:012]: 'Virtual Screen' Turning ON.
[13:00:54][D][sensor:094]: 'Desk Height': Sending state 62.70000 cm with 1 decimals of accuracy
[13:00:54][D][script:077]: Script 'screen_timer' restarting (mode: restart)
[13:00:54][D][main:322]: Executing screen timer command
[13:00:54][D][switch:012]: 'Virtual Screen' Turning ON.
[13:00:54][D][sensor:094]: 'Desk Height': Sending state 62.80000 cm with 1 decimals of accuracy
[13:00:54][D][script:077]: Script 'screen_timer' restarting (mode: restart)
[13:00:54][D][main:322]: Executing screen timer command
[13:00:54][D][switch:012]: 'Virtual Screen' Turning ON.
[13:00:55][D][sensor:094]: 'Desk Height': Sending state 63.40000 cm with 1 decimals of accuracy
[13:00:55][D][script:077]: Script 'screen_timer' restarting (mode: restart)
[13:00:55][D][main:322]: Executing screen timer command
[13:00:55][D][switch:012]: 'Virtual Screen' Turning ON.
[13:00:55][D][sensor:094]: 'Desk Height': Sending state 63.60000 cm with 1 decimals of accuracy
[13:00:55][D][script:077]: Script 'screen_timer' restarting (mode: restart)
[13:00:55][D][main:322]: Executing screen timer command
[13:00:55][D][switch:012]: 'Virtual Screen' Turning ON.
[13:00:55][D][sensor:094]: 'Desk Height': Sending state 63.70000 cm with 1 decimals of accuracy
[13:00:55][D][script:077]: Script 'screen_timer' restarting (mode: restart)
[13:00:55][D][main:322]: Executing screen timer command
[13:00:55][D][switch:012]: 'Virtual Screen' Turning ON.
[13:00:56][D][sensor:094]: 'Desk Height': Sending state 64.20000 cm with 1 decimals of accuracy

IamMattM, 2023-05-26:
When pressing the "up" arrow on the keypad I only get to see the following messages:
[13:03:02][D][sensor:094]: 'Desk command': Sending state 1.00000  with 0 decimals of accuracy
[13:03:02][D][sensor:094]: 'Desk command': Sending state 8.00000  with 0 decimals of accuracy

IamMattM, 2023-05-26:
Enabling the logger baud rate to 9600 had definitely helped in revealing what might be going on...

Scoff123, 2023-05-26:
Ok great that's really helpful, we are narrowing the issue down at least. So:

1) Keypad commands are being sent to the ESP board, and are successfully received and decoded by ESPHome.
2) *PROBLEM* ESPHome is not transmitting commands which originate from the keypad, onto the controller.

3) Home Assistant commands are being sent to the ESP board successfully.
4) HA commands are successfully being transmitted onto the controller and processed correctly.

Hopefully someone with a bit more ESPHome knowledge can maybe identify what could be going on here with the onward transmission logic of the 'keypad received' commands?

IamMattM, 2023-05-26:
An update:

I memorised Preset 1+2+3 using the standard setup (no ESP32 man-in-the-middle) and once using the ESP32 back in the path between keypad and eq5, all of those Presets 1+2+3 are working from the keypad !!

The UP/DOWN key control do not work but then again it does not look like the code interpret those commands correctly as "Up" and "Down" - 

This has been done without first triggering any controls from the HA integration just to be sure....

IamMattM, 2023-05-26:
Since disabling the "Logger" section is seems to be more robust and has not played up when using the Preset1+2+3

Scoff123, 2023-05-26:
Getting closer! So I wonder what is different between the up/down key press logic and the preset button press logic? 
I'll see if I can find what condition triggers the rST message to display, if that information is contained in a troubleshooting guide for the desk/controller anywhere online, or maybe from FlexiSpot themselves. 

I don't have anything connected up yet, I've simply flashed my ESP32 Devkit board with the yaml file in this repo, and edited slightly for the eq5 naming, however I DO have the stop button in my list of controls after adding my ESP32 board as an integration in HA:

[image]

so I'm not sure why you don't have that. If you go into Dev Tools in HA and services, do you have the service cover.stop_cover available?

IamMattM, 2023-05-26:
Hmmm. That is most pecular...
I'll have a look.

IamMattM, 2023-05-26:
Sorted ! Appears to be me tweaking the file for those (open/close/stop) statements.

I reverted back and althouth there is a warning during compilation, it does get through and ca nbe installed on ESP32.
_______________________________________________________________________________________________________________________________________
Compiling .pioenvs/esp32-flexispot-eq5/src/main.cpp.o
/config/esp32-flexispot-eq5.yaml: In lambda function:
/config/esp32-flexispot-eq5.yaml:130:22: warning: 'void esphome::cover::Cover::open()' is deprecated: open() is deprecated, use make_call().set_command_open() instead. [-Wdeprecated-declarations]
                 id(desk).open();
                      ^
In file included from src/esphome/core/controller.h:14,
                 from src/esphome/components/api/api_server.h:4,
                 from src/esphome/components/api/api_connection.h:6,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/components/cover/cover.h:133:8: note: declared here
   void open();
_________________________________________________________________________________
 
        
I did that and up/down keys now working !

Success.

Scoff123, 2023-05-26:
Well done :) Good result there. I will revert my code back to the original as well in that case, for the open/close/stop. 

Are you still getting any of the rST messages displaying on the keypad LCD, or have they stopped now? I was wondering if that is something to do with the height going out of bounds or something like that, and so the controller required a soft recalibration by putting the legs all the way down maybe?

If you manage to add the preset4 to the yaml, would you mind posting it in here as well please? 

if there's anything I can check or test in regards to your stop button not appearing in HA, I'm happy to help - just let me know!

IamMattM, 2023-05-26:
I got Preset4 on EQ5 keypad to work from HA.

However, supporting it from keypad is a bit more channlenging for my knowledge.
The keypad.h only checks the fourth byte but it should check both fourth and fifth byte to be able to distinguish Preset4 from the Empty command - Both have byte four at 0x00

[image]

IamMattM, 2023-05-26:
> Well done :) Good result there. I will revert my code back to the original as well in that case, for the open/close/stop.
>
> Are you still getting any of the rST messages displaying on the keypad LCD, [...]

Sure. Will share the three files and post a video to show EQ5 is supported reasonably well.

Scoff123, 2023-05-26:
I've just tried compiling using the newer lambda format of the ESPHome code (https://esphome.io/components/cover/index.html), so that I'm no longer using the deprecated version:
```
id(desk).open(); => id(desk).make_call().set_command_open().perform();
id(desk).close(); => id(desk).make_call().set_command_close().perform();
id(desk).stop(); => id(desk).make_call().set_command_stop().perform();
```
I'm no longer getting the warnings during compile, and have successfully flashed to my ESP32 board. I can't however test this, as I have nothing wired up yet. Would you mind updating your yaml to include this revised version, and checking to see if the up and down buttons on the keypad still work correctly?

Scoff123, 2023-05-26:
Great, thanks for testing that - glad it works and has removed the deprecated warnings during compile.

So it sounds like we are almost there and fully compatible. Hopefully we can figure out a revised keypad header file to incorporate the check of bytes 4 and 5. 

I'm just waiting on my keystone Jacks to arrive so I can wire up my board and test and up/down lockup issue, given I have the stop button present in HA. We can then rule that out as a cause if it still happens.

I take it you've tried removing your eq5 integration from your esphone integrations in HA, and then re-adding it? Can't think why else the standard stop button would be missing from a cover, unless it was a glitch while adding it in HA 🤔

IamMattM, 2023-05-26:
> e it you've tried removing your eq5 integration from your esphone integrations in HA, and then re-adding it? Can't think why else the standard stop button would be

Yes, indeed. Almost there. Not too bad an outcome.
Would be great if someone knowledgeable could adapt the code to decode byte 4+5 to support Preset4 from keypad.
I can leave without it but it would be great to sort it out nevertheless....

I'll post a couple of pics of the wiring between ESP32 and the two keystone RJ45 jacks in the 3D printed enclosure.

I did remove and reload the entire ESPHOME id and gave it a new name but that cover stop button is not showing.Odd.

I will have a look at other covers that I have integrated (for example a garage door cover that shows the stop button)

david-romero, 2023-06-08:
Hi @IamMattM!

Thank you so much for your research and great work.
I'd like to implement this approach and the first thing I need is to buy the hardware.
I think I need the following hardware:

- 2 RJ45 Sockets: [Amazon](https://www.amazon.es/sourcingmap%C2%AE-Modular-Conector-Conectores-Acopladores/dp/B01FLDTFBK/ref=d_pb_allspark_dp_sims_pao_desktop_session_based_sccl_3_4/257-2864433-5416151?pd_rd_w=h4mlz&content-id=amzn1.sym.7939c1d9-0489-4c74-9518-f95ac9c8a9c7&pf_rd_p=7939c1d9-0489-4c74-9518-f95ac9c8a9c7&pf_rd_r=AK976B0ND37AYH6RCE5J&pd_rd_wg=Ff6NE&pd_rd_r=198a1546-76ae-4a3a-969a-102e016667e7&pd_rd_i=B01FLDTFBK&psc=1)
- ESP32: [Amazon](https://amzn.eu/d/inZCmUh)
- Wiring: [Amazon](https://amzn.eu/d/5PZ9O0p)

Am I right?
Thank you in advance!

IamMattM, 2023-06-08:
> Hi @IamMattM!
>
> Thank you so much for your research and great work. I'd like to implement this approach and the first thing I need is to buy the hardware. I think I need the following hardware: [...]

It might be more convenient to use/buy the following ? :
1) Keystone RJ45 sockets (pack of x5 , Euro 9.99)   [https://www.amazon.es/VCE-Paquete-Keystone-Ethernet-Punch-Down/dp/B07P4LMDP6/ref=sr_1_5?__mk_es_ES=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=1FGGIXD1M3U5J&keywords=keystone+rj45&qid=1686242120&s=industrial&sprefix=keystone+rj45%2Cindustrial%2C69&sr=1-5](url)

- I did not have a punch tool but the following wire stripper worked a treat because of the notch:
[image]

You might already have this as part of cheap soldering iron kits available like this one (Euro 18.69): 
[https://www.amazon.es/Ockered-Temperatura-interruptor-mantenimientos-electrodom%C3%A9sticos/dp/B07WYLF6QV/ref=sr_1_1_sspa?__mk_es_ES=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=100H4RHJ77LNE&keywords=soldering+iron&qid=1686242620&refinements=p_36%3A2493682031&rnid=2493681031&s=tools&sprefix=soldering+iron%2Caps%2C84&sr=1-1-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9hdGY&psc=1](url)

2) ESP32  - Yep, the one you pointed to would work. 

3) Wires (free from old ethernet short cable ?): any standard ethernet cable that you can sacrifice , should already have the necessary wires inside with the benefit of the colour coding already matching the keystone Mode B wiring colours.

david-romero, 2023-06-09:
Thank you so much @IamMattM.

So the shopping list would be:

- Keystone RJ45 sockets (pack of x5 , Euro 9.99) [Amazon RJ45 keystone](https://www.amazon.es/VCE-Paquete-Keystone-Ethernet-Punch-Down/dp/B07P4LMDP6/ref=sr_1_5?__mk_es_ES=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=1FGGIXD1M3U5J&keywords=keystone+rj45&qid=1686242120&s=industrial&sprefix=keystone+rj45%2Cindustrial%2C69&sr=1-5)
- ESP32: [Amazon](https://amzn.eu/d/inZCmUh)

I have some old ethernet cables. 😄 

Have I had to solder something? Why do you suggest buying a soldering kit? Does the RJ45 keystone need to be soldered?

IamMattM, 2023-06-09:
@david-romero 

David, I suggested the soldering kit  because 
1) You get this wire stripper tool that can be used to "push" the ethernet wires into the keystone (no soldering at this end) (otherwise you would need t oacquire a tool called a punch tool )

2) You also get a soldering iron+solder to solder the other end of those wires onto the actual ESP32 pin header (you might even need to solder the ESP32 pin header if it does not come pre-assembled....)

Your initial choice of RJ45 connectors would have more than likely also required soldering even with socket-to-socket dupont wires (not that you were looking at socket_to_pin header version of those dupont wires)

Scoff123, 2023-06-09:
I'm going to order the 3D Printed enclosure online as i can't stretch to buying a printer right now, but I bought a pack of 5 VCE Keystone jacks like the ones that @IamMattM linked above, and have miles of spare ethernet cable :) I'll be doing the wiring once I have the enclosure and fitted the keystone jacks to get the lengths correct. I take it there were no changes to any of the wiring / colour diagrams posted above? Your google drive pictures seem to indicate not, but thought I better check before I start connecting things together.

IamMattM, 2023-06-09:
> 

@Scoff123 

No changes since posting those pictures of the wiring in the G.drive.

Scoff123, 2023-06-09:
@IamMattM  Great thanks for confirming 👍

iMicknl, 2023-06-10:
@IamMattM (and others) would you be willing to contribute your knowledge and designs to this repository? So that people don't have to scroll through all comments in this issue.

Your work looks great and this will help many people further.

IamMattM, 2023-06-15:
@iMicknl 
Credit to you for sharing this great repo.

You are wellcome to use any of the posted information above for the repo.
[google drive](https://drive.google.com/drive/folders/1MgeNHxeG5XQShUc6OGoT1LHJDNu98LrI?usp=drive_link
)

The "summary" is that the EQ5 can be supported with the ESP32 acting a the "man-in-the-middle" and presenting two RJ45 socket.

The 3D printable enclosure and the RJ45 blank place in the google drive above is courtesy of others .

The EQ5 yaml is a working one and the photos show hopefully enough resolution to make sense of the connections required.

Only EQ5.keypad position#4 is not functional when operated from physical keypad (needs firmware change to filter on both byte#4+5)

THanks again for having shared your work - I now have two FLexispot EQ5 "wired" up - 

Note:

deveritt-uk, 2023-06-22:
Firstly huge kudo and credit to both of you @IamMattM and @Scoff123, all of your work above has been absolutely amazing! It was a great read.

I came across your above thread the other day, but needed to order some more ESP32's as using all my others around the house already. 

I have got keystones already (pretty geeky) and plenty of jumper leads / ethernet, also a 3D printer for the case.... 

What I dont have though is a clear wiring diagram, you said to iMicknl "while the photos will hopefully show enough resolution for others to make sense of the connections required." Unfortunately I can not make sense of them as I'm partially colour blind... (rofl)... 

I was wondering if there was any way of maybe putting a basic sketch together (in your paint of choice) if at all possible?

IamMattM, 2023-06-23:
@deveritt-uk : 

Try "FlexiSPot_EQ5_Connections.pdf" in the Google repo above.
[image]

IamMattM, 2023-06-23:
Also please note that if I was to do it again I would flip the DEVKIT as to being able to see the connection names before soldering the wires ( I already had the module headers soldered and did not want to have to un-solder them for this ...)

deveritt-uk, 2023-06-23:
Huge thanks @IamMattM! 

That's really great! I thought a few of them were doubled up, so will have to get the soldering iron out to look in to this, I had my partner sat with me trying to decode the colours last night (ha) but we couldn't quite follow them all the way around haha... (Plus was getting late for us)

Shall give this a try out in the next few days and let you know! 

Thanks again!

deveritt-uk, 2023-06-27:
Evening, so I have it fully wired up and flashed the controller as shown below:
[image]
[image]
[image]

 I have it flipped as you mentioned @IamMattM, but used the schematic so it "should" be fine... 

When I have it all hooked up to my EQ5, the screen read out shows 68.1 and I can pull that through to Home Assistant in to my dashboard as 68.1 as shown below:
[image]

 but... I can't send any commands and HA also can not push commands through. 

Just wondering if there is anything I can to try and "trouble shoot" my wiring? I suspect something isn't wired up correctly, possibly my TX2

Thanks in advanced!

IamMattM, 2023-06-27:
@deveritt-uk 

I think that there is something pecular with the connection at the Keystone end: it appears that you have blue/white-blue going in the keystone spot for brown/white-brown ?

IamMattM, 2023-06-27:
I don't suppose you might have inadvertently reversed which keystone support what ?
Like, could you have perhaps swapped which port does what ? That could have the outcome you described.

Try swapping the RJ45 leads on the outside of the box or please double check the keypad goes to the left hand side as viwed on your top-view capture.

IamMattM, 2023-06-27:
When compiling the code for ESP32, did you also make use of my versions of the  .yaml and with the two .h files ? All in the G.drive.

Those files would already have been adapted to take the VirtScreen (aka keypad is present) to a default=ACTIVE for it to work straight away.

IamMattM, 2023-06-27:
Just looked at your wiring and it is mixing things up...:

Example:
If I look at <IO23=Blue_EQ5> it correctly goes towards the EQ5 port showing on the right hand side.
But If I try to follow <IO3-green_kpd> is also goes towards the EQ5 port on the right hand when it should have been going to the left-hand side one.(keypad port)

deveritt-uk, 2023-06-27:
@IamMattM 

Thanks for the above pointers and info, 

It turned out that I had switched the Green 6 and had them the wrong way around so pins 17 and IO3 where mixed up. Checked all the other wiring on the Keystones and they still matched the schematic, I guess that's what you get for trying to multi task at the same time as wiring things up... 

Swopped them over and now it's working a treat! 

Virtual Screen is indeed active by default as yes I had compiled from the latest yaml in the google drive and also used the two .h files in the esphome conf section.

Thanks for all the tips and pointers

deveritt-uk, 2023-06-27:
> I think I found your issue:
> 
> If I look at <IO23=Blue_EQ5> it correctly goes towards the EQ5 port showing on the right hand side.
> 
> However, If I try to follow <RX0/IO3=green_kpd> it is also incorrectly going towards the EQ5 port on the right hand side when it should be going to the keypad port on the left hand side - This is why it is not working and is easily fixed by :
> 
> **swapping out the two internal green wires connections at the ESP32 end, to make it good.**
> 
> The rest of the internal wires look ok to me. And of course, externally attach to the intended port EQ5 port (right) towards EQ5 box, while Keypad port (left) attaches to keypad.

Just seen your reply above as I was typing out, many thanks!

IamMattM, 2023-06-27:
One more thing: the WiFi antenna could be upset by the presence of the metal screw head where you have it in your pics.
I would consider relocating towards USB connector if there is no clash.

This is only to maximise your wireless range once fitted underneath the desk.

[45 more replies]