iMicknl/LoctekMotion_IoT #37: Support request for model HCB213A-1

lboue, 2022-10-03:
Hello,

The HCB213A-1 model does not appear on the supported models page.
Here is a picture below:

[image]

Regards

horriblenoises, 2022-10-21:
I just got an [EN1B desk frame off amazon](https://www.amazon.com/dp/B07GWCSXJ8) (in the US) with this model controller and connected a D1 mini to it with the ESPhome instructions and the e5b yaml file. It required minimal modifications (min/max height adjustment & adding preset 4) but I'm very happy to say it seems to work really well. Cover up/down travel the desk to min/max, the presets all work, and the height readout is correctly displayed in Home Assistant.
[image]

renjithstalin, 2023-01-24:
@horriblenoises Can you please share your preset 4 entry? also, does down button work for you? Thanks in advance

horriblenoises, 2023-01-25:
@renjithstalin  

Sorry. I got distracted from this project for a bit. Never added the problems I'd found after posting that. It seems like the up/down buttons don't work after a timeout occurs. Pin20 doesn't seem to wake it either. So I changed up/down to internal in the esphome yaml. My D1 mini also reboot a lot after a command is finished. I have to investigate if it's something I've done or something that isn't compatible.

This aside, it does work as well as I need it to. The preset buttons are not effected by the timeout issue so I set P1 to min, P4 to max, and if I need granularity beyond that, I can always add an automation to Home Assistant or the ESP to call for specified heights (send button press P4; if height = height requested -1, send button press P4 again to stop).

```
  - platform: uart
    name: "Preset 1"
    id: ${device_name}_preset1
    icon: mdi:numeric-1-box
    data: [0x9b, 0x06, 0x02, 0x04, 0x00, 0xac, 0xa3, 0x9d]
    uart_id: desk_uart

  - platform: uart
    name: "Preset 2"
    id: ${device_name}_preset2
    icon: mdi:numeric-2-box
    data: [0x9b, 0x06, 0x02, 0x08, 0x00, 0xac, 0xa6, 0x9d]
    uart_id: desk_uart

  - platform: uart
    name: "Preset 3"
    id: ${device_name}_preset3
    icon: mdi:numeric-3-box
    data: [0x9b, 0x06, 0x02, 0x10, 0x00, 0xac, 0xac, 0x9d]
    uart_id: desk_uart

  - platform: uart
    name: "Preset 4"
    id: ${device_name}_preset4
    icon: mdi:numeric-4-box
    data: [0x9b, 0x06, 0x02, 0x00, 0x01, 0xac, 0x60, 0x9d]
    uart_id: desk_uart

  - platform: uart
    name: "Up"
    id: ${device_name}_up
    icon: mdi:arrow-up-bold
    data: [0x9b, 0x06, 0x02, 0x01, 0x00, 0xfc, 0xa0, 0x9d]
    uart_id: desk_uart
    internal: true

  - platform: uart
    name: "Down"
    id: ${device_name}_down
    icon: mdi:arrow-down-bold
    data: [0x9b, 0x06, 0x02, 0x02, 0x00, 0x0c, 0xa0, 0x9d]
    uart_id: desk_uart
    internal: true
```

renjithstalin, 2023-01-25:
Many thanks for the response. I have managed to get Preset 4 working. But my down button still doesn't work. I'm going to try this with esp32 and see if it works. And yes, my wemos d1 mini reboots too :(
Thanks again.

sebPomme, 2023-02-07:
> It seems like the up/down buttons don't work after a timeout occurs.

I read somewhere on another model that UP/DOWN work only if the panel light are turn on before :thinking: so the trick was to send an impulsion to a pin that are link to the screen before doing an action .  
I c'ant remember where.

May we can apply the same pattern.

lourdrivera123, 2023-05-02:
> Many thanks for the response. I have managed to get Preset 4 working. But my down button still doesn't work. I'm going to try this with esp32 and see if it works. And yes, my wemos d1 mini reboots too :( Thanks again.

Were you able to test it on esp32? 
We have the same model of control panel, but I don't know how to map it correctly with my RJ45 and ESP32 Devkit V1.

ThomDietrich, 2023-10-06:
Hey guys, thanks for confirming that it works for the HCB213A-1. I bought the desk "Sanodesk EC4 "EC4B-EU and it came with the controller box HCB2**2**3A-1. I expect it to be identical, except for the two-motor-drive. I'll try with the e5b yaml and an esp32 the next couple of days.

For who might find it useful, here are some pictures of the controller:

[image]
[image]
[image]

ThomDietrich, 2023-10-06:
Couldn't wait 😄 

I've quickly thrown together an RJ45 to ESP32 cable and flashed the firmware. I can confirm that everything is working. This includes the up and down commands - no wake-up or PIN20 needed (both of those seem to have no effect on my specific device).

The only bug I found is that "Desk Height" stays unknown after the ESP boots up. This can be resolved via a wake-up (display or programmatically) and isn't a bigger issue for my use cases. Still, I found another workaround for waking up the device: If a preset button is not programmed to a specific height, it will wake up the display but won't move the desk.

iMicknl, 2023-10-06:
@ThomDietrich would you be willing to test https://github.com/iMicknl/LoctekMotion_IoT/issues/67? Hopefully this will also solve your issues.

Zagur, 2024-01-18:
> Couldn't wait 😄
>
> I've quickly thrown together an RJ45 to ESP32 cable and flashed the firmware. I can confirm that everything is working. 🎉 This includes the up and down commands - no wake-up or [...]

I currently have an HCB213A-1 and I was testing with the configuration for e5b with what you have mentioned. I am using an ESP328266 Mod, what scheme should I follow to connect the cables correctly?

ThomDietrich, 2024-01-18:
Hey @Zagur, you should find everything you need in https://github.com/iMicknl/LoctekMotion_IoT/issues/67

Zagur, 2024-01-19:
First of all I want to thank everyone for this incredible project.

I have been reading several issues and have configured my ESP32 2866 to control the HCB213A-1 controller. Everything seems to be working correctly except for one thing I've seen in Home Assistant.

When I try to set the height to 90cm it always adds between 2 and 2.7cm extra. I use the mushroom-number-card to show the height but it doesn't work very well. A value is always set below or above what is indicated.

Doing some tests I get some ideas/doubts:

- Would it be possible that the switch_up and switch_down, instead of going up to the maximum or minimum, did it in 5 at a time or 10 at a time? (maybe put a parameter to configure this).

Regards!