iMicknl/LoctekMotion_IoT #42: One command - 2cm change

KamilKobylarek, 2022-12-12:
Hi. I have a problem because I can't control my desk well. Sending one command over uart changes the height by about 2cm which is not the case with a regular controller. Is there any way to make the control more precise?

sickill, 2023-01-21:
Hey Kamil, what desk do you have, and which controller box?

KamilKobylarek, 2023-01-21:
HS13A-1 and CB38M2L(IB)-1

sickill, 2023-01-22:
I have CB38M2L(IB)-9 and HS13B-1 and I noticed something similar: when I raise/lower the desk from HA via cover controls whenever I click stop it takes a moment (~0.5s) for it to notice it and stop. This results in desk stopping a few cm lower/higher than I wanted it.

naresh97, 2023-02-04:
> Hi. I have a problem because I can't control my desk well. Sending one command over uart changes the height by about 2cm which is not the case with a regular controller. Is there any way to make the control more precise?

On my CB38M2J and HS13B-1, sending the wakeup command stops the desk movement. So you can control the magnitude of the change by varying the delay between a move and wakeup command.