iMicknl/LoctekMotion_IoT #47: Question about control box cb28m1j(ib)-1 and HS13A-1

Lucavon, 2023-02-11:
Hey!
I have a HS13A-1 control panel with a cb28m1j(ib)-1 control box (according to a sticker on the side). I have tried getting this to work, but haven't had any success, none of the commands issues by home assistant resulted in any action (I used ESPHome).
I believe that the wiring might be incorrect, even though I connected the wires just as listed here: https://github.com/iMicknl/LoctekMotion_IoT/tree/main/packages/esphome 

How can I find out what the correct wiring is for my combination of control box and panel?
Thanks!

lufetico, 2024-07-26:
Is your desktop a Flexispot E8? From the data you indicate I believe it is. Have you been able to solve the problem? I have not been able to get the cabling to work.

Lucavon, 2024-07-26:
I don't know what exact model I have, but I have since given up after not being able to make ot work. I'm sure there's a way, but I didn't have the time to try.