iMicknl/LoctekMotion_IoT #49: Commands dont execute unless i wake desk from control panel CB38M2L(IB)-1 and HS01B-1

briodan, 2023-02-17:
i'm running this on an ESP8266 D1 mini and i'm running into an issue where the commands don't execute unless I wake the desk from a physical control button. After waking the desk from a physical button i get less then 10 seconds to execute a command from the ESP before it goes unresponsive

The ESP8266 is connected via the second RJ45 port, while the HS01B-1 panel is connected to the first port.

here is my config

```
substitutions:
  device_description: ESP8266 Custom
  friendly_name: desk
  name: desk
  min_height: "62.6" # Min height + 0.1
  max_height: "127.9" # Max height - 0.1

esphome:
  name: ${name}
  comment: ${device_description}
  platform: ESP8266 
  board: d1_mini
  includes:
    - desk_height_sensor.h
  on_boot:
    priority: -10
    then:
      - switch.turn_on: switch_m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_key

ota:
  safe_mode: true
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${friendly_name} Fallback Hotspot"
    password: !secret fallback_password

captive_portal:

web_server:
  port: 80 

uart:
  - id: desk_uart
    baud_rate: 9600
    tx_pin: D5
    rx_pin: D6

sensor:
  - platform: wifi_signal
    name: "${friendly_name} WiFi Signal"
    update_interval: 60s

  - platform: uptime
    name: Uptime

  - platform: custom
    lambda: |-
      auto desk_height_sensor = new DeskHeightSensor(id(desk_uart));
      App.register_component(desk_height_sensor);
      return {desk_height_sensor};
    sensors:
      id: "desk_height"
      name: Desk Height
      unit_of_measurement: cm
      accuracy_decimals: 1
      icon: "mdi:counter"

button:
  - platform: restart
    name: "${friendly_name} Restart"   

switch:
  - platform: gpio
    name: "Virtual Screen"
    pin:
      number: D2
      mode: OUTPUT
    restore_mode: ALWAYS_ON
    internal: true

  - platform: uart
    name: "${friendly_name} Preset 1"
    id: switch_preset1
    icon: mdi:numeric-1-box
    data: [0x9b, 0x06, 0x02, 0x04, 0x00, 0xac, 0xa3, 0x9d]
    uart_id: desk_uart

  - platform: uart
    name: "${friendly_name} Preset 2"
    id: switch_preset2
    icon: mdi:numeric-2-box
    data: [0x9b, 0x06, 0x02, 0x08, 0x00, 0xac, 0xa6, 0x9d]
    uart_id: desk_uart

  - platform: uart
    name: "${friendly_name} Preset 3"
    id: switch_preset3
    icon: mdi:numeric-3-box
    data: [0x9b, 0x06, 0x02, 0x10, 0x00, 0xac, 0xac, 0x9d]
    uart_id: desk_uart

  - platform: uart
    name: "${friendly_name} Up"
    id: switch_up
    icon: mdi:arrow-up-bold
    data: [0x9b, 0x06, 0x02, 0x01, 0x00, 0xfc, 0xa0, 0x9d]
    uart_id: desk_uart
    internal: true

  - platform: uart
    name: "${friendly_name} Down"
    id: switch_down
    icon: mdi:arrow-down-bold
    data: [0x9b, 0x06, 0x02, 0x02, 0x00, 0x0c, 0xa0, 0x9d]
    uart_id: desk_uart
    internal: true

  - platform: uart
    name: "${friendly_name} M"
    id: switch_m
    icon: mdi:alpha-m-circle
    data: [0x9b, 0x06, 0x02, 0x20, 0x00, 0xac, 0xb8, 0x9d]
    uart_id: desk_uart

  - platform: uart
    name: "${friendly_name} (wake up)" # Not available on all control panels
    id: switch_wake_up
    icon: mdi:gesture-tap-button
    data: [0x9b, 0x06, 0x02, 0x00, 0x00, 0x6c, 0xa1, 0x9d]
    uart_id: desk_uart

cover:
  - platform: template
    # icon: mdi:table-chair
    # icon: mdi-human-male-height-variant
    name: "${friendly_name}"
    assumed_state: true

    # Move desk up
    open_action:
      - while:
          condition:
            sensor.in_range:
              id: desk_height
              below: ${max_height}
          then:
          - logger.log: "Executing up command"
          - switch.turn_on: switch_up
          - delay: 10ms
    
    # Move desk down
    close_action:
      - while:
          condition:
            sensor.in_range:
              id: desk_height
              above: ${min_height} 
          then:
          - logger.log: "Executing down command"
          - switch.turn_on: switch_down
          - delay: 10ms
    optimistic: true
```

log sample from the logs of executing a command and nothing happening
```
15:33:43 | [D] | [sensor:126] | 'desk WiFi Signal': Sending state -59.00000 dBm with 0 decimals of accuracy
15:33:46 | [D] | [switch:013] | 'desk Preset 3' Turning ON.
15:33:46 | [D] | [switch:056] | 'desk Preset 3': Sending state ON
15:33:46 | [D] | [uart.switch:020] | 'desk Preset 3': Sending data...
15:33:46 | [D] | [switch:056] | 'desk Preset 3': Sending state OFF
15:33:56 | [D] | [sensor:126] | 'Uptime': Sending state 423.09698 s with 0 decimals of accuracy
15:34:43 | [D] | [sensor:126] | 'desk WiFi Signal': Sending state -60.00000 dBm with 0 decimals of accuracy
15:34:56 | [D] | [sensor:126] | 'Uptime': Sending state 483.09698 s with 0 decimals of accuracy
15:35:43 | [D] | [sensor:126] | 'desk WiFi Signal': Sending state -59.00000 dBm with 0 decimals of accuracy
15:35:56 | [D] | [sensor:126] | 'Uptime': Sending state 543.09802 s with 0 decimals of accuracy
15:36:43 | [D] | [sensor:126] | 'desk WiFi Signal': Sending state -60.00000 dBm with 0 decimals of accuracy
15:36:56 | [D] | [sensor:126] | 'Uptime': Sending state 603.09698 s with 0 decimals of accuracy
15:37:43 | [D] | [sensor:126] | 'desk WiFi Signal': Sending state -60.00000 dBm with 0 decimals of accuracy
15:37:56 | [D] | [sensor:126] | 'Uptime': Sending state 663.09900 s with 0 decimals of accuracy
15:38:43 | [D] | [sensor:126] | 'desk WiFi Signal': Sending state -59.00000 dBm with 0 decimals of accuracy
```

iMicknl, 2023-02-18:
Did you follow the instructions and did you connect and turn PIN20 on or off?

briodan, 2023-02-19:
ok now i feel like and idiot turns out there was a loose connection on one of the pins that would slip into place everytime i pushed the button on the control panel.

closed 2023-02-19 by briodan