iMicknl/LoctekMotion_IoT #58: FlexiSpot E7 Pin Map in Readme

bryanhunwardsen, 2023-06-20:
I created 2 ethernet harnesses to map FlexisSpot E7 to UpsyDesky.
I was using the pin map contained in root readme here (Control Box HS13B-1)
Along with the ethernet pin/color identification resources contained.

It **appears** the E7 pin map (Pin numbering column) is incorrectly inverted.
Looking at my panel plug (clip up, plugging in away from me) Pin 1 is Brown, Pin 8 is Yellow, Pin8 is 5v and pin 7 is gnd -validated with meter) see pic

[image]

bryanhunwardsen, 2023-06-20:
Looks as pin numbers for all the maps are inverted.
This shows plug upside down from my pic, but indicates more so the pin id #'s need to be inverted in your readme.

[image]

bryanhunwardsen, 2023-06-20:
Sorry to many wires and cables upside down. Everything is correct spologies.

CoderTobi, 2023-10-31:
I agree with you. The table in the readme file seems to be inverted or mirrored. 
I used a RJ45 Breakoutboard to connect the ESP to the table and used a multimeter to determine which pin is 5V.

The highlighted collum shows the correct pin config. The one right next to it is how its documented in the readme-file.
[image]

Since the table is apparently using 5V on the UART pins this is very unhealthy for the ESP as the TX and RX pins don't match in the documentation from the readme file.

jamesmyatt, 2024-09-23:
I think I fixed this is #91 on the v2 branch.

Jonas354, 2024-10-01:
Yes, same pin-out.