iMicknl/LoctekMotion_IoT #61: Anti-Collision function not present.

sumit-anantwar, 2023-08-24:
I have a LocTek Motion desk with the `CB28M1B-1` controller.
The Instruction Manual mentions Anti-Collision as an Optional feature, but it seems to be absent in my setup.
Did anyone find a way to enable/configure this?

iMicknl, 2023-09-19:
Hold up and down button for 5 seconds. You should see a response A-1, A-2 or A-3 which is the sensitivity.

sumit-anantwar, 2023-11-02:
@iMicknl Thank you for the reply.
I have tried this, and it does not do anything.
It seems the controller does not support this feature.
I tried contacting LoctekMotion, but didn't receive any answer from them.

So, I was wondering if there's any hack that I can apply to enable Collision Detection.