iMicknl/LoctekMotion_IoT #62: ESP won't boot up with keypad connected

formatBCE, 2023-09-08:
Maybe someone already had this problem:
My device works really good, everything is supported. I power it from 5v cord of the table controller.
However, if i pull the table plug out and put it back in, ESP won't boot, if keypad was connected to RJ45 socket...
So i need to pull keypad out of the device, reset ESP and put it back again after 5-6 seconds. And even then, sometimes ESP blinks and stops responding - today, for example, it took me 4 tries to put it back to work with keypad connected.

Is it something related to the power? I checked all connected pins, and they shouldn't prevent ESP from starting, according to docs...

formatBCE, 2023-10-11:
So from my findongs, "Pin20" contact has 5v (actually 4.7) on it. Not safe to use directly with ESP32 GPIO (although might work for some time, but can eventually break the chip). I tried to build small voltage divider on that, but it doesn't work as well.
At current time i'm out of ideas, and can't understand why i'm the only one with this trouble.

If i don't connect Pin20 from keypad to ESP, it boots up normally - but nothing works, operations don't take any effect.

I tried 2 different boards, with as clean config as i could: NodeMCU ESP32, and ESP32-MINI boards. Have same results on both.

BCould it be, that my desk keypad is somehow different? It's Flexispot E5.

formatBCE, 2023-10-11:
Here's full thread of my lurkings - i actually tried a lot of stuff:
https://www.reddit.com/r/esp32/comments/1742yoa/comment/k4anb0z/

iMicknl, 2023-10-11:
@formatBCE what is your pin-out? Have a look at https://github.com/iMicknl/LoctekMotion_IoT/issues/67.

In theory most ESP32's should be tolerant to 5V.

formatBCE, 2023-10-11:
@iMicknl according to data sheets, they aren't. Max allowed voltage on GPIO is 3.6-3.9V depending on board, if the specific board has no special voltage handling included. By docs, it MIGHT work for some time, but is dangerous long-term.

My pinout is according to that Reddit post, here's the image: https://www.reddit.com/media?url=https%3A%2F%2Fi.redd.it%2Fy5d1kf1ar8tb1.png

I tried different pins yesterday for both UART and PIN20, behaviour is unchanged.

iMicknl, 2023-10-11:
@formatBCE the pinout on your image is hard to read. Should I read it as pin 1 to 8 (from top to bottom) on your desk controller and keypad? Please have a look at my link and try the other pin-out.

Regarding 5V tolerance, I have to say this is quite a mess. There are not many official sources. https://hackaday.com/2016/07/28/ask-hackaday-is-the-esp8266-5v-tolerant/ describes a bit (with sources), but no official statement. In my own tests, I had issues with ESP8266 and with ESP32 it has been solved.

formatBCE, 2023-10-11:
Color			What	Desk->ESP		ESP->Control		Desk->Control

Brown			VCC		VCC 							+
White-brown		GND	GND							+
White-blue		RX		GPIO16							+
Green			TX		GPIO17			GPIO26
Blue				Pin20	GPIO23			GPIO22

Maybe this is better? All other colors from controller are going directly to keypad.
It's actually default pinout. Also i tried with GPIO18 instead of GPIO26, and GPIO27 instead of GPIO22. Same behaviour.

formatBCE, 2023-10-11:
@iMicknl updated my comment. Freaking Github and its markdown :)

iMicknl, 2023-10-11:
Haha, makes sense. If I am not mistaken, your pin-out is still different than the one mentioned in https://github.com/iMicknl/LoctekMotion_IoT/issues/67? Or did you start counting in reverse?

Have you tried with my config, without your desk keypad connected? So no pass through.

formatBCE, 2023-10-11:
@iMicknl you're right, i started in reverse... Looked on wrong table in README. yeah it's 8->7->...->4.
Will try with no passthrough now.

iMicknl, 2023-10-11:
Thanks! And please try with my new YAML as well. I have been running this for a long time and it is stable, especially compared to previous versions.

formatBCE, 2023-10-11:
@iMicknl tried with your config and replaced sensor file code just in case.
It works, without keypad. (Not mentioning some weirdness because my desk height is in inches, and some buttons are screwed too.

iMicknl, 2023-10-11:
Thanks! What is different for a desk that uses inches? Is changing the numbers (and unit just for visual purposes) not enough?

formatBCE, 2023-10-11:
Yeah, min/max height, units - that's it :)

formatBCE, 2023-10-11:
@iMicknl frankly saying, it wasn't giving me troubles to rule the desk without passthrough before. Just needed to set PIN20 to always on - (virtual screen that is).

But the best thing would be to have keypad also working. :)

formatBCE, 2023-10-24:
Alright, i unplugged pin20 cable, that is going from keypad to ESP, and, using your new code from #67 it works now, no hanging on reboot. Will try it a bit more, but looks like it's working. I guess i will have to live without child lock, but it's alright.