iMicknl/LoctekMotion_IoT #64: D1 mini seems to crash when power is supplied via Ethernet [HS13B-1, Flexispot E7]

j3mu5, 2023-09-18:
Hello all,

Thanks to the great documentation, I was able to connect my table [HS13B-1, Flexispot E7] with HomeAssistant. However, the D1 mini seems to restart occasionally - it loses the connection with HomeAssistant (buttons are grey). After a few seconds, the D1 restarts and connects with HomeAssistant and works as desired again. This is also evident in the uptime (screenshot). I suspect that the power supply through the table is not stable. Is this already known to be a problem?

[image]

If I do not connect the D1 mini to the table and power it via USB, it runs stably for 30,000 seconds (approx. 8 h).

iMicknl, 2023-09-19:
@j3mu5 I am working on improved documentation at the moment, and most likely the recommendation will be to use an ESP32 chip. Most desks provide 5V, where a ESP8266 works with 3.3V. You could use a 5V to 3.3V stepdown in between if you use ESP8266. Additional benefit of using ESP32 is that it has hardware UART.

I haven't been able to fully test my statement above, but this is most likely the cause. I will test this and update the documentation.

j3mu5, 2023-09-19:
@iMicknl  Thank you for the explanation. In the meantime, I tried to supply the power separately via USB and only establish the data connection via Ethernet. The table could be controlled, but unfortunately I observed crashes again.

Only when the d1 Mini is operated without connection to the table do I see no restarts.

Never mind - I follow your advice once the documentation is done! I happen to have ordered wemos esp32 recently, for espresense. One of these can then also used for the desk. Or is there another ESP 32 board that can be recommended?

Thanks for your work on this project! I'm really looking forward to the table moving to my height when HA recognises me at the desk! 🤩

iMicknl, 2023-09-19:
The documentation in the README is a bit outdated, and there are some improvements to the YAML and C++ code to be made. I will integrate the work of all great contributors on GitHub + Discord and update the documentation accordingly.

niggowai, 2025-01-04:
I had the same problem with an esp8266 board and using software UART for communication.
The board apparently hang up and was resetted by the watchdog.

When I changed the logger component to log on `UART1` and my pins to `tx_pin: GPIO01` and `rx_pin: GPIO03` to be able to use hardware `UART0`, the board ran stable.

Drawback here is you have to resolder for flashing, but since esp8266 supports ota this might be acceptable.