iMicknl/LoctekMotion_IoT #66: HS13M - probably works the same

SaiCode-DEV, 2023-09-27:
I bought the Flexispot E7 Pro and it has the HS12M remothe. is there a simple way to test/check if the protocol changed?

iMicknl, 2023-09-27:
Not really. But if you have a RJ45 port on your controller, big chance protocol is supported. You could also open your display/remote and see what the pinout is.

SaiCode-DEV, 2023-09-27:
[image]

Sadly no spesific Information under the cables

iMicknl, 2023-10-06:
@SaiCode-DEV the easiest way to test this is to just hook up an ESP32 to your control box. See https://github.com/iMicknl/LoctekMotion_IoT/issues/67.

MauiR, 2024-04-08:
Hello, I also own the Flexispot E7 Pro desk. Unfortunately, with the new Office-desk.yaml configuration in ESPHome (https://github.com/iMicknl/LoctekMotion_IoT/blob/6a1da5dc22688cd4eb8ca8e90e5c4b123c4354d9/packages/office-desk.yaml), only the desk height functionality is working. All other features such as Alarm, Desk position, Preset 1, Preset 2, Sit, and Stand are non-functional.
The "desk height" however, is working perfectly and updates in real-time

I've utilized an ESP32 with this RJ45 connector layout: https://github.com/iMicknl/LoctekMotion_IoT?tab=readme-ov-file#hs01b-1

[image]

iMicknl, 2024-04-08:
@MauiR have you tried the new configuration and pin-out from https://github.com/iMicknl/LoctekMotion_IoT/issues/67?

MauiR, 2024-04-08:
> @MauiR have you tried the new configuration and pin-out from #67?

Yes, I've chosen the new configuration pinout. However, I haven't connected SWIM and RES because there were no specifications provided for them on the ESP32.

MauiR, 2024-04-20:
I contacted FlexiSpot to inquire if they have a new protocol and if they could provide information about it. Unfortunately, they weren't very cooperative and were unwilling to assist.

MauiR, 2024-05-20:
I can say that I have now managed to get it running with the Flexispot 7 Pro. However, it only works with an ESP8266. I haven't been able to get it running with an ESP32. I'll take another closer look.

Rudd-O, 2024-07-29:
I'm interested in this too.  Pinout and software.  How can I help?

locnh, 2025-06-10:
I have an `E7Q` with `HS13M` keypad, work smoothly with these boards:
- `esp32s3-devkitc-1`
- `esp32c6-devkitc-1`
- `esp32-mini-d1`

closed 2026-10-28 by iMicknl