iMicknl/LoctekMotion_IoT #67: [Testers wanted] New ESPHome configuration and documentation improvements

iMicknl, 2023-10-01:
I am working on improving this repository. The main goal is to make it easier to get started, by providing only two configurations and a more updated README + configuration. In order to achieve this, all feedback and contributions are welcome. 

**Main goal:**
- [ ] Only provide 2 options and pin-outs (LoctekMotion Desk and LoctekMotion Desk with pass through)
- [ ] Advise to use ESP32 due to 5V tolerance and hardware UART, add note about ESP8266 with voltage shifter (5V -> 3.3V).
- [ ] Simplify the README and installation instructions (add a version for non technical people, 'plug and play')
- [x] Modernize YAML config with cover, buttons and number entity
- [ ] Remove the need for `desk_height_sensor.h` (load from custom component)

**Long term**:
- [ ] Use ESPWebtools (https://esphome.github.io/esp-web-tools/) to make it easier to get started

## Changes
- Switch will turn off automatically when min or max height is reached
- New cover entity 
- New number entity (still buggy)
- Presets are exposed as button entities
- Experimental buttons for alarm / alarm off

## Pin-out
In the past we have discussed multiple pin-outs, but it seems that multiple pin-outs can be supported by the same desk. 

| RJ45 pin | Name      |  ESP32 |
| -------- | --------- | --------- | 
| 8        | +5V (VDD) |  VIN |
| 7        | GND       | GND |
| 6        | TX        | TX2 (GPIO17) |
| 5        | RX        |  RX2 (GPIO16) |
| 4        | PIN 20    | D23 (GPIO23) |
| 3        | (unknown) |  |
| 2        | SWIM      | |
| 1        | RES       | |

You can change the ESP32 pin-out mapping, but don't forget to change the pin number in the YAML.

## Configuration
[office-desk.yaml](https://github.com/iMicknl/LoctekMotion_IoT/blob/v2/packages/office-desk.yaml)
[desk_height_sensor.h](https://github.com/iMicknl/LoctekMotion_IoT/blob/v2/packages/desk_height_sensor.h)

The following secrets should be part of your ESPHome config: `encryption_key`, `wifi_ssid` and `wifi_password`.

## Feedback
Any feedback and contributions are welcome! For example the work above adopted for pass through configurations.

ThomDietrich, 2023-10-06:
Hey @iMicknl,

thanks for providing this updated version of the code. The improvements look really good but I did barely test them. First observations:

- You have mixed up preset 3 and 4 in the comments. On my device this is Button 4: https://github.com/iMicknl/LoctekMotion_IoT/blob/6303fa7d10fc1cc102740a4a41e9dff448cd5c42/packages/office-desk.yaml#L158
- On my specific device the wake screen command and the PIN20 method do not work, however, "child lock" (which seems not supported) wakes up the device (and produces the current desk height) - so thank you!
- `update_interval: 0ms # update in every loop() iteration` in number "Desk Height" does what you intended, but is that really necessary?

General comment: It seems like more and more devices will be potentially supported by your project, and each will have their own caveats. My HCB223A-1 seems to be a good example. Should we capture those differences in a table of sorts?

brumouta, 2023-10-07:
Where one gets a board like that one you showed (rj45 connector with pins)?

Thanks

iMicknl, 2023-10-08:
Thanks a lot for your extensive feedback, @ThomDietrich! 

> On my specific device the wake screen command and the PIN20 method do not work, however, "child lock" (which seems not supported) wakes up the device (and produces the current desk height) - so thank you!

The wake command does not work here as well. What do you mean with PIN20 doesn't work? PIN20 is set to `ALWAYS_ON` (high) to allow the ESP32 to control the desk.

>update_interval: 0ms # update in every loop() iteration in number "Desk Height" does what you intended, but is that really necessary?
This floods the log indeed, and I need to have a better look here. It needs to be at least 0.5s, in order for the number template to be updated with the height of the desk height sensor.

> A preset button will send the desk on its way. This can be stopped by e.g. shortly pressing the Up or Down button. Could you add a "Stop" button?
Good one! Perhaps we could make the cover.stop button behave like this. I am not sure how often you want to cancel a preset command?

> A binary_sensor that indicates that the desk is currently moving would be great
In theory the `cover` entity should have such a status already. We need to have a look if this is currently missing and if so, if this could be added.

> You've used the name "Desk Height" twice. Yes not a technical issue, more of an intuitive ick

Do you have any suggestions for a better name / structure? The issue here is that one is a sensor and the other one is a number entity (that is still a bit buggy).

ThomDietrich, 2023-10-09:
> PIN20 is set to ALWAYS_ON (high) to allow the ESP32 to control the desk.

Oh! I didn't know that. Compare here: https://github.com/iMicknl/LoctekMotion_IoT#execute-a-command
Maybe you want to add a description to the README? I wouldn't know because on my device the pin does not have a purpose, i.e., `ALWAYS_OFF` does not change any behavior.

> It needs to be at least 0.5s, in order for the number template to be updated with the height of the desk height sensor.

I get that. My first thought was a delta/throttle filter but glad I tested it: Filters are only supported by sensors. One option i can think of is to disable direct updates and do the following:

```yaml
      id: "desk_height"
      on_value:
        then:
          - component.update: set_desk_height
```

I've tested the solution and it works as expected.

>  I am not sure how often you want to cancel a preset command?

Probably very seldom but to me it feels like this action is missing from the options offered in HA. I would expect the Stop button to stop all movements, be it preset or cover actions. Yes the cover has this function but I generally like to offer slightly redundant but purposeful options. Makes integration in e.g. automations or a Lovelace dashboard a lot easier.

> In theory the cover entity should have such a status already. We need to have a look if this is currently missing and if so, if this could be added.

I thought so as well but doesn't seem like it 🧐 Anyhow, for the same reasons as above, I'd vote for a dedicated binary_sensor.

>  Do you have any suggestions for a better name / structure?

Good challenge. Nope can't think of one. Let's leave as is.

jamesmyatt, 2023-10-09:
BTW, ESP8266 is also tolerant to 5V inputs. e.g. https://www.tablix.org/~avian/blog/archives/2022/05/a_footnote_about_esp8266_5v_tolerance/

iMicknl, 2023-10-09:
> Oh! I didn't know that. Compare here: https://github.com/iMicknl/LoctekMotion_IoT#execute-a-command
Maybe you want to add a description to the README? I wouldn't know because on my device the pin does not have a purpose, i.e., ALWAYS_OFF does not change any behavior.

Interesting. So in your case you don't have PIN20 connected to your ESP + no PIN20 in your config?

Thanks for your other improvements, I will give it a try and update the startpost when I am done.

>BTW, ESP8266 GPIO pins are also tolerant to 5V inputs. e.g. https://www.tablix.org/~avian/blog/archives/2022/05/a_footnote_about_esp8266_5v_tolerance/

Thanks, @jamesmyatt! Not sure why I had so many stability issues when using the ESP8266 with the same config. Perhaps the combination of 5V + no hardware UART causes these stability issues. Multiple users reported similar behavior with ESP8266, but I wasn't able to pin point the issue.

ThomDietrich, 2023-10-09:
> Interesting. So in your case you don't have PIN20 connected to your ESP + no PIN20 in your config?

Historically motivated I have it connected and configured as internal switch as per your default config. However I can confirm that the switch does not have any effect, i.e. neither `ALWAYS_OFF` and `ALWAYS_ON`, as well as switching during runtime will interrupt normal operation.

Imho, let's simply state that PIN20 shall be connected and configure it as `ALWAYS_ON`. I'd suggest a different name other than "Virtual Screen", which I can't relate to. How about "Remote Control Pull-up"?

ThomDietrich, 2023-10-10:
Minor additional comment: In your code some `name` and `id` settings are quoted, not all. None of them need quotes.

Slightly tangent but maybe relevant for docs? As I am using an [ESP32-C3](https://docs.m5stack.com/en/core/stamp_c3u) I used GPIO21 as TX, GPIO20 as RX.

iMicknl, 2023-10-10:
Thanks @ThomDietrich, I will make the changes. And actually hope to soon have a beta branch where everyone can easily contribute.

j3mu5, 2023-10-15:
Thanks @iMicknl  for the rework! With the new code, ESP Home runs very stable on an ESP32 chip! So far I haven't noticed any more crashes. Good work! 

I still observed two features that don't quite match the expected behaviour.

1) If the height is selected via HA using the slider, the desk always moves a little too far, i.e. to slightly too high positions when going up and to slightly too low positions when going down.

3) If the 4 seat positions are selected via HA (Preset 1, Preset 2, Sit, Stand), the desk height is not updated.
This can only be done manually if
a) I press the button "Wake Screen" while the display is still on.
b) when the display is off, one presses the "Memory" button.

iMicknl, 2023-10-15:
Thanks for the feedback, @j3mu5!

1. This is a known issue indeed, and won't be too easy to solve. Other implementations keep retrying (so going up and down) until they reach the requested height. I see this feature more as a gimmick anyways, since most desks have 4 presets.
2. Which desk do you have? For me, the height updates as well when I set a preset.
3. This will probably be changed to a 'push' mechanism as suggested by @ThomDietrich.

j3mu5, 2023-10-15:
Dear @iMicknl , it is an honour for me to contribute by testing and giving feedback! :)

1. I also don't find it essential that the table hits the exact height when set by the slider. As you say, this works perfectly with the presets. I don't find repeated adjustment with up and down (like other implementations) an improvement. 
I have one further idea: How about subtracting (upwards movement) or adding (downwards movement) a certain, user selectable offset (1-2 cm) from the target size. This would at least compensate for the error somewhat. The difference between the target and the actual value is unfortunately variable; between 1 and 2 cm in my case.

2. It is the Flexispot E7 with a Wemos D1 mini 32. I have connected the cables as follows:
GND White-Brown
+5V (VDD) Brown
GPIO17 Green
GPIO16 White-Blue
GPIO23 Blue

3. Cool, thanks @ThomDietrich

brumouta, 2023-10-15:
My Flexispot E7 also don't update height when clicking presets

iMicknl, 2023-10-15:
@brumouta do you use the pin-out and config from above?

brumouta, 2023-10-15:
Im using an esp8266 (lolin nodemcu), with the pins from the main branch but this new yaml.

iMicknl, 2023-10-15:
@brumouta if you would be able to use the pin-out from this branch (if it isn't the same), would be great. It should work for your device as well.

brumouta, 2023-10-15:
It should be moving the rx and tx to gpio13 and gpio 15 and leaving pin20 where it is?
[image]

iMicknl, 2023-10-15:
I meant the pin-out on the side of your desk mainly, doesn't matter much how you link it to your ESP. The original approach has multiple options for pin outs on the desk side.

JannikSchmuck, 2023-10-15:
Hey @iMicknl 

thanks for the updated version of the code, it made it so much easier for me to implement the project at my desk. 

As I'm a total newbie with ESPs and HA I could assist in improving the documentation if needed, just let me know if I can help with that, by the way the reference to the ESPWebtools made the implementation a lot easier for me.

The issue with the desk height not being updated when using the presets as @j3mu5 mentioned I can confirm, this also occurs on my E8 (HS13B-1) with ESP32 Dev.

brumouta, 2023-10-15:
@iMicknl yeah, i'm using the same pins that @j3mu5 is using. I can try tomorrow the new config, and the number from 1 to 8 is basically following the image from main branch right? Or it's just inverted here, because it's basically the same thing from the opposite side.

ThomDietrich, 2023-10-16:
@iMicknl what's stopping the merge of v2 branch into main? Seems like it already is a big improvement, even if further improvements can come through individual discussions as we had them here

iMicknl, 2023-10-16:
> @iMicknl yeah, i'm using the same pins that @j3mu5 is using. I can try tomorrow the new config, and the number from 1 to 8 is basically following the image from main branch right? Or it's just inverted here, because it's basically the same thing from the opposite side.

@brumouta, it is the same as https://github.com/iMicknl/LoctekMotion_IoT?tab=readme-ov-file#hs01b-1 from the main branch. The main branch has different pinouts, and it seems this specific pin-out works on all devices.

l7ssha, 2023-10-19:
This version of esphome config works much better than the ones fro repo. I modified to add keypad passthrough and it seems to work really well. I used .h file and sensor snippets from repo - probably not the best way of implementing it, but works nonetheless. My desk was bought from elzap.eu with HS13B-1 controller.

https://gist.github.com/l7ssha/c6d989396a560f895ee3c770815a650a

brumouta, 2023-10-20:
> @brumouta, it is the same as https://github.com/iMicknl/LoctekMotion_IoT?tab=readme-ov-file#hs01b-1 from the main branch. The main branch has different pinouts, and it seems this specific pin-out [...]

Wait, this is one is the same as the other one based on RJ45 colours, but the pins order are inverted, now I'm more confused

formatBCE, 2023-10-24:
Works good (also with pass-through). One thing i noticed is that ESP is showing height logs each loop cycle. Hopefully it doesn't try to update sensor in HA with this frequency, but only on change.

iMicknl, 2023-10-24:
@formatBCE this is something that I still need to change, currently it spams the log.

tobit6795, 2023-10-28:
> This version of esphome config works much better than the ones fro repo. I modified to add keypad passthrough and it seems to work really well. I used .h file and sensor snippets from repo - [...]

can you share your wiring? i have CB28M1B-1 + HS01B-1  but cant get it to work with passtrough

formatBCE, 2023-10-28:
> > This version of esphome config works much better than the ones fro repo. I modified to add keypad passthrough and it seems to work really well. I used .h file and sensor snippets from repo - probably not the best way of implementing it, but works nonetheless. My desk was bought from elzap.eu with HS13B-1 controller.
> > 
> > https://gist.github.com/l7ssha/c6d989396a560f895ee3c770815a650a
> 
> @l7ssha 
> 
> can you share your wiring? i have CB28M1B-1 + HS01B-1  but cant get it to work with passtrough

WDYM, code in that gist works.

iMicknl, 2023-10-28:
@formatBCE my initial post here only explains the wiring for non pass through. I think that @tobit6795 is looking for an example of the wiring when you use pass through.

CoderTobi, 2023-10-31:
Hi, sorry to bust into this issue like this.

I started out with the [flexispot_ek5.yaml](https://github.com/iMicknl/LoctekMotion_IoT/blob/main/packages/esphome/flexispot_ek5.yaml) example, modified it to work with my Flexispot E7 and am now adopting the changes from your code. So far, I have run into some issues and had some questions:

In the current readme-file there seems to be an error concerning the pin map of the Flexispot E7. It looks like the pin map is inverted, like @brumouta already mentioned. See here for more information: #58 
@iMicknl Is this expected when using a RJ45 breakout board? 
Additionally I think it would be a lot better to list the pin number of the RJ45 instead of the color of the wires as they aren't really useful when using a breakout board and can also be different from cable to cable. Furthermore, I think adding another collum with the labels of the motor controller (not the control panel) would make sense and prevent mixing up of TX and RX pins.

In other words, I would change this (Pins mirrored on E7):

ESP8266 | Loctek Motion (HS13B-1) | Loctek Motion Cable Color (HS13B-1) | Ethernet cable color (T568B)
-- | -- | -- | --
GND | GND | Blue | White-Brown
D6 | RX | Green | White-Blue
D5 | TX | Black | Green
D2 | PIN 20 | Red | Blue

Into this (pins corrected for E7):

Motor controller CB38M2J(IB)-1 | Control panel HS13B-1 | Loctek Motion Cable Color (HS13B-1) | Ethernet cable number | ESP32
-- | -- | -- | -- | --
GND | GND | Blue | 7 | GND
TX | RX | Green | 5 | GPIO 16 (RX)
RX | TX | Black | 6 | GPIO 17 (TX)
Pin 20 | PIN 20 | Red | 4 | GPIO 23
+5V | +5V | ? | 8 | Voltage regulator (if available)

Note: I don't know if the colors from the Loctek Motion Cable are also inverted for the E7

Another thing that has already been mentioned is the usage of a logic level shifter. While the EPS32 apparently can handle a 5V input, I doubt it's going to be healthy. So a logic level shifter would definitely make sense. The problem is, which one? It would be nice if we could make a table or something like that, listing the experiences of people who already used them. For example, is a I2C logic level shifter able to convert the signals fast enough?

Additionally, do we actually know that the motor controller uses 5V for UART? My multimeter isn't capable of verifying that, as it  constantly shows 2V between TX and GND on the controller. Can somebody verify that it's actually using 5V. Possibly someone with an oscilloscope? I'd rather not fry the controller of the desk than the ESP32...

stonegray, 2023-11-03:
Is it possible to limit the update rate? I'm getting about 50 messages/s to my mqtt server when nothing is happening. 

I tried unsuccessfully to update `set_desk_height` whenever `desk_height` changes, but can't seem to get it to work.

formatBCE, 2023-11-03:
> Is it possible to limit the update rate? I'm getting about 50 messages/s to my mqtt server when nothing is happening. 
> 
> I tried unsuccessfully to update `set_desk_height` whenever `desk_height` changes, but can't seem to get it to work. 

Yeah I guess author said it should be fixed. Cached somewhere inside or so. Because it bombs with updates

stonegray, 2023-11-03:
- Can't stop the desk using the cover control when moving using a preset; works fine if movement was started using the cover control.

- Can't move the desk to a preset if starting from the maximum height. To reproduce, press up and wait until desk hits top, then use a preset. 

- I'm still getting long floats (eg. `99.099998`) in the number component on HA even though I see the `step`/`accuracy_decimals` are set.

Using MQTT instead of Esphome API, desk with a CB28M1B-1 control box.

iMicknl, 2023-11-03:
For now, you can remove the number entity and your logs won't be flooded. The update interval there is the issue. Need to update this push, instead of pull.

ThomDietrich, 2023-11-03:
> time to finish the instructions and polish everything! 😁.

Hey @iMicknl, your call but I personally believe that you should just merge as is and additional changes, like to readme can be addressed step by step. Didn't we already resolve the number updates issue?Let's merge :)

iMicknl, 2023-11-06:
@ThomDietrich agree, however the README isn't fully updated yet. Main blocker was to find some time to do this, but hopefully I can do it this week. Just changed the YAML linked above to include the `component.update` to no flood the logs.

I am still not sure why multiple users face the following:
> If the 4 seat positions are selected via HA (Preset 1, Preset 2, Sit, Stand), the desk height is not updated.
This works like a charm for me with the YAML + wiring above. (without passthrough).

> Can't stop the desk using the cover control when moving using a preset; works fine if movement was started using the cover control.
This is normal behavior. You can cancel your movement by selecting the same, or another, preset again.

ThomDietrich, 2023-11-07:
Hey @iMicknl, regarding the `component.update` change to the number entity. I just realized that you can change the `update_interval: 0ms` to a literal `update_interval: never  # will be updated by another component`. Pretty neat. Thought you might wanna add that

1100101, 2023-12-15:
Hi,

I've been trying to get this system to work, but with limited success.

I have a CB38M2J(IB)-1 control box and a (HS13A-1) touch panel.
By control box has 2 RJ45 ports, one is used by the touch panel, the other is now connected to the ESP.

I use an ESP8266, with the pinout described above.
Since I don't have HomeAssistant, I added MQTT support by just adding these lines:
```
mqtt:
  broker: 192.168.178.57
  client_id: esp_desk
```

stonegray, 2023-12-15:
IIRC there's a line you can comment out to see the raw uart data. 

Is this issue specifically related to the version here or does it also occur with the main version?

1100101, 2023-12-16:
I only tried with this new "v2" version of the ESPHome script, I haven't tried the version from the main branch.

hugo-clemente, 2024-01-03:
Hey there, I've been trying to get this working on my e7q for about an hour now, without success. I figured, what the hell, might as well try to unplug/plug the desk, and it finally worked. Maybe this could be added in the documentation as a silly troubleshooting step 😅

1100101, 2024-01-04:
I have a feeling that although my control box has 2 RJ45 ports, the one with the ESP doesn't receive any updates done by the key panel connected to the other port.

I tried this: With the ESP connected, I just removed PIN20, then changed the desk height via key panel, and reconnected PIN20 to the ESP. Almost immediately, the new height was published via MQTT.

iMicknl, 2024-01-04:
> I have a feeling that although my control box has 2 RJ45 ports, the one with the ESP doesn't receive any updates done by the key panel connected to the other port.

I have seen multiple reports on this, but I can't reproduce this.. My control box has 2 RJ45 ports and when I manually change the height via the keypad, I see the values change via my ESP32.

Related to your testing with pulling the PIN20. Have you tried turning the PIN20 on/off via the ESP?

1100101, 2024-01-04:
@iMicknl How would I do that?
(This is my very first ESP, and I'm not at all familiar with ESPHome).

ThomDietrich, 2024-01-04:
Look for this part of the code:

```
switch:
  - platform: gpio
    name: Serial Control Pull-Up  # "PIN20" of desk controller
    pin:
      number: GPIO9
      mode: OUTPUT
    restore_mode: ALWAYS_ON
    entity_category: config
    internal: true
```
Change the last line to `internal: false`. You will see a new switch turn up in Home Assistant.

1100101, 2024-01-04:
I don't have Home Assistant, but I added an MQTT command to toggle the PIN20 switch.  
If i toggle the switch off for at least 2 seconds, and then on again, the height data packages are being processed (I enabled the debug logging in `height_sensor.h`), and if the hight has changed it is being published.
But if I don't "touch" the virtual switch, and change the height of the table, no debug messages regarding height are being processed.

voydz, 2024-02-15:
Thanks for this project. I really appreciate the work that has been done.

Just as a note. I had to download the height sensor C implementation and use it with 'includes:' as the 'external_components:' entry does not work for me. (compile error)
Am I missing something? 

I have the same issues with the reporting of the desk height. (Two RJ45 ports that seem to report different things.) 
Will do some testing and share my findings if I got something interesting. 

best regards

fbloemhof, 2024-03-09:
Seems to work fine!

My control box is a "CB38M3B(IB)-1". I found out my ethernet wiring is the other way around, but this might be my own fault. I only have RX/TX connected now and all looks good. If I use the original controller it also updates the esp status.

I am unable to power the ESP32 via the desk for now, might be this issue: https://github.com/esphome/issues/issues/4864. Will look into this later. 

How can I add a preset and is it possible to adjust the alarm (timer)?

Also I would like to change the behavior of the up and down buttons to "momentary", the same as the desk itself. Would this be possible?

gschmidl, 2024-03-12:
I tried this on my standing desk with the HS13A-1 controller and a second RJ45 port and it worked instantly - what took the longest was getting the ESP32 into flash mode.

jamesmyatt, 2024-03-22:
Is this the same as the version in the v2.1 branch, which looks like it's the latest? https://github.com/iMicknl/LoctekMotion_IoT/tree/v2.1

markuspi, 2024-03-27:
Using my own esphome configuration, i got my Flexispot E7 stuck in `ASR` and `E07`  errors. Resolving the problem via esphome did not work right away: Continuously sending the _move down_ command using an interval of 100ms resulted in slow, jerky motion and did not fix the error. Only after i adjusted the `send_every`  interval down to 50ms, i was able to move the table down normally and fix the error

bobtheshizocat, 2024-03-31:
Thanks for the repository @iMicknl, and the helpful comments from everyone to read through!

I have a Flexispot E7 with the control box `CB38M2E(IB)-1` and the control panel `HS13B-1`. I'm using an `ESP32-WROOM-32` dev board (esp32dev) for the smart addon. This is my wiring (using the same one mentioned [here](#issuecomment-1763439839)):

| ESP32         | Cable-Color | RJ45-PIN |
|---------------|-------------|----------|
| +5V (VIN)     | Brown       | 8        |
| GND           | White-Brown | 7        |
| GPIO16 (RX2)  | White-Blue  | 5        |
| GPIO23        | Blue        | 4        |
| GPIO17 (TX2)  | Green       | 2        |
<details>
  <summary>Click here to expand for an image of the wiring</summary>
  [image]
</details>

**Does work as expected:**
- [x] Desk (Up- / Down- / Stopbutton has a short delay)
- [x] Alarm
- [x] Preset 1
- [x] Preset 2
- [x] Sit
- [x] Stand
- [x] Desk Height (Sensor)
- [x] Alarm off
- [x] Child Lock
- [x] Memory
- [x] Restart

**Doesn't work as expected:**
- [ ] Desk Height (Number): Shows the current height, but if I enter a value, it sets the desk either too high while going up or sets it too low while going down.
- [ ] Wake Screen: Doesn't work.

dominik-b13, 2024-04-02:
> Thanks for this project. I really appreciate the work that has been done.
>
> Just as a note. I had to download the height sensor C implementation and use it with 'includes:' as the [...]

i solved the compile error by using an older esphome version (2023.6.5 worked for me, didnt test any newer)

Qu4ndo, 2024-04-17:
I have a **Flexispot E7 Pro** and built it using a **NodeMCUv2 Board (ESP8266)**.
The pin-out worked as expected but I had to lookup my specific wiring diagram for the used ethernet cables (B standard).
After also changing the pins to the correct ones on my NodeMCUv2 board (including changing it in the esphome yaml) everything worked right out of the box!

**NodeMCUv2 Mapping:**

| RJ45 pin | Name | ESP32 |
| ------------- | ------------- |------------- |
| 8 | +5V (VDD) | VIN |
| 7 | GND | GND |
| 6 | TX | TX (GPIO1) |
| 5 | RX | RX (GPIO3) |
| 4 | PIN 20 | D7 (GPIO13) |
| 3 | (unknown) | |
| 2 | SWIM| |
| 1 | RES| |
<br>

**Additional Notes:**
- **Don't forget to place the file "desk_height_sensor.h" into your esphome config base directory (ssh/sftp/terminal)!**
- The ethernet cable ends have twisted pairs (in my situation "T-568B" standard) but only the pin number is important.
- My Desk Controller is the model **"CB38M2L(IB)-2"** with the Panel **"HS13M-1C0"** 
<br>

**Testing:**

| Function | Default | Working |
| ------------- | ------------- |------------- |
| Height Button Control <br> (Up/Down/Stop) | true | true |
| Alarm | false | false |
| Sit | true | true |
| Stand | true | true |
| Preset 1 | true | true |
| Preset 2 | true | true |
| Display Desk Height | true | true |
| Alarm Off | false | false |
| Child Lock | false | false |
| Memory <br> (plus Button press afterward) | true | true |
| Restart | true | true |
| Wake Screen | false | false |

*Note:* 
- "Default" means if it is a function the table should have out of the box.
- "Height Button Control" works in ~5cm steps. In order to get only one step you have to instantly press the stop button.
- "Wake Screen" works automatically with the press of a preset button, but not manually.

jamesmyatt, 2024-04-20:
With ESPHome 2024.4.0, I get the following compile error:

```
/config/esphome/eh-086b7c.yaml: In lambda function:
/config/esphome/eh-086b7c.yaml:104:37: error: expected type-specifier before 'DeskHeightSensor'
  104 |       auto desk_height_sensor = new DeskHeightSensor(id(desk_uart));
      |                                     ^~~~~~~~~~~~~~~~
/config/esphome/eh-086b7c.yaml:106:33: error: could not convert '{desk_height_sensor}' from '<brace-enclosed initializer list>' to 'std::vector<esphome::sensor::Sensor*>'
  106 |       return {desk_height_sensor};
      |                                 ^
      |                                 |
      |                                 <brace-enclosed initializer list>
```

The desk height sensor and external_components are as in the v2.1 branch. The only significant difference is that I'm using the ESP8266 platform rather than the ESP32. I don't know if this matters.

jamesmyatt, 2024-04-20:
OK. It seems to work well when `desk_height_sensor.h` is added via the `esphome/includes` entry. But not via the external components. So I think that the v2.1 config is inconsistent in the way it's using the desk_height_sensor. I'll make a PR to fix this.

voydz, 2024-04-24:
> OK. It seems to work well when `desk_height_sensor.h` is added via the `esphome/includes` entry. But not via the external components. So I think that the v2.1 config is inconsistent in the way it's using the desk_height_sensor. I'll make a PR to fix this.

Yeah, think so, too.

Now, on ESPHome 2024.4.1, it will not even compile with a local `desk_height_sensor.h`. It is a bit outdated, I think it will need an update. Unfortunately I did not look into ESPHome too much.

jamesmyatt, 2024-05-23:
PR #83 is to revert to using the "custom component" local include logic rather than the external component. This works.

However, PR #84 instead implements the sensor as a proper external component and can be tested by using the yaml config from the PR with the following modification:

```
external_components:
  source: github://jamesmyatt/LoctekMotion_IoT@v2.2
  components: [ loctekmotion_desk_height ]
```

iMicknl, 2024-05-28:
Thanks, @jamesmyatt. I will dive into this. Just returned from a long holiday, so will try to review in the coming week. 😊

exico91, 2024-06-04:
Hello!
I just bought a Flexispot E7 and I have the controller HS13M-1C0
Im using a ESP32-C3 mini from Wemos (im using the version 1.0.0, it needs a tweak in the wifi section, *note1) https://www.wemos.cc/en/latest/c3/c3_mini_1_0_0.html

I followed the pinout from Qu4ndo https://github.com/iMicknl/LoctekMotion_IoT/issues/67#issuecomment-2061667090

The control box has two RJ45 connectors, I used one for the panel and one for the ESP instead of using a passtrough.

| **RF45 pin** | **Name**  | **ESP32-C3** |
|--------------|-----------|--------------|
| 8            | +5V       | VBUS         |
| 7            | GND       | GND          |
| 6            | TX        | 4            |
| 5            | RX        | 5            |
| 4            | PIN 20    | 10           |
| 3            | (unknown) |              |
| 2            | SWIM      |              |
| 1            | RES       |              |

Obviously place the "desk_height_sensor.h" file in the same directory as the yaml.

(Up/Down/Stop) - Works
Sit - Works, its preset 4
Stand - Works, its preset 3
Preset 1 - Works
Preset 2 - Works
Desk height - Works
Alarm - Not found in the yaml for E7
Alarm Off - Not found in the yaml for E7
Child Lock - Not found in the yaml for E7 *note2
M - Works
Wake Screen - not working, maybe because Im using the second RJ45 port instead of a passtrough?

*note1: 
If you use a C3 Mini V 1.0.0 set in the wifi section "output_power: 8.5"
There is a HW bug and the solution is to limit the wifi power, otherwise it works

*note2:
There is a code for the child lock or is just a M key spam?

*extra:
I also have bluetooth proxy enabled on this ESP because I have devices in the home that use bluetooth and  why not using one that is hidden and forever on? Works no problem.

[8 more replies]

closed 2024-12-13 by iMicknl