iMicknl/LoctekMotion_IoT #73: FlexiSpot Q8 "Comhar Pro" w/ controller HCB215C-4

mcomer-captech, 2023-12-11:
I just bought this desk and would like to implement some control / automation. This controller is a single unit with motor controls and display integrated.

[image]

Internally the primary PCB is identified as HCB115C-1.
 
It does have a single RJ45 along with power in, two motors, and a DC-out for wireless charging and a board-edge connector which has a cover on it:

[image]
[image]
[image]
[image]

Any thoughts on this one? I have an oscilloscope w/ login analyzer and can do some research and digging here on my own also. I did create a little breakout cable from the RJ45 and did identify ground and +5v, but everything else appears to be floating or ground with a little bit of switching / motor noise. I suspect I may need to do something to "wake up" this interface, e.g. the "pull a pin high" (pin 20 thing) and/or wake-up command I've seen referenced elsewhere - but hard to do that without some kind of pinout.

I would love some suggestions on how to get started here and will of course share all of my findings back with the community.

iMicknl, 2023-12-11:
Thanks for your extensive post! You could try starting with the following pin-out: https://github.com/iMicknl/LoctekMotion_IoT/issues/67.

mcomer-captech, 2023-12-12:
@iMicknl unfortunately no luck yet with the info from #67. I do get +5v on Pin 1 and Gnd on Pin 2.

I threw together a sketch to send commands from an ESP32 I have around. I tried using both the "wake" as well as the "child lock" commands (since someone else said their controller seemed to respond to this command). I tried sending these both with and without "PIN20" pulled high. I used my 'scope to monitor several of the other pins for activity. Nothing I did caused the screen to light up or the controller to respond in any way, nothing on any of the other lines. I do see some slight voltage shifts and some motor controller noise on several of the other lines if I manually activate the controls.

This feels very much like the situation discussed in #25. I don't see that the gents there ever made any progress? It feels like there is something needed to activate this port. I may have to resort to following traces on the board and trying to figure out what goes where.

One other option - the manufacturer did very helpfully label  SDA and SCL on the little interconnect that goes between the main board and the display / button board - so maybe some kind of I2C man-in-the-middle? Probably very complicated to implement...

madninjaskillz, 2024-03-26:
I have one of these and have it somewhat working. The "universal" pinout doesn't work, i think i used the HS13B-1 wiring. It seems the serial data has changed slightly on this model:

* Up + Down work but the frequency of 108ms seems to be less than how long an up/down command executes for.
* Presets do not work
* pulsing memory will update height (which can often get out of sync)
* pulsing memory will turn on screen - but sometimes needs to be pulsed twice.
* Virtual Screen doesnt seem to do anything

because up and down operate for a while, stop is unreliable and setting a height via a number value is unusable (often overshoots)

While commands are being sent to the desk, the control panel on desk is unresponsive, and currently the only sure way to stop an UP is to send DOWN, things get jammed. 

I'm currently very unscientifically trying stuff out via modifying the esp home yaml, might try and get the esp home to act as a serial converter so i can be a bit more methodical.