iMicknl/LoctekMotion_IoT #80: ESP8266 D1 Mini not receiving output

philxws692, 2024-03-03:
Hey there, 

I stumbled upon this great project and gave it a try. 
I use a Wemos D1 Mini with the ESP8266 chipset as a controller, unfortunately I had to go through a little bit of trial and error because the desk I got, which is the Flexispot E6, with the HS17B-1 handset, was not confirmed to be working.
Nevertheless I somehow managed to get it sending commands to the desk and using my presets. However it does not receive any output from the controller, I already confirmed this by using the `ESP_LOGD` function from the `desk_height_sensor.h`.
For the RX Pin I tried using the RX Pin on the Wemos D1 Mini, and also tried using something like the D5, D6 Pins as described, but I can not get it to work.

Does anybody had similar experience, maybe also with the Flexispot E6?

Nama, 2025-01-15:
Should have looked for issues for this... I'm struggling with this issue since yesterday and I am out of ideas...

I'm using the ESP8266 NodeMCU dev board.

<details>

<summary>My code</summary>

```yml
ota:
  - platform: esphome
    password: "xxxxxxxx"

substitutions:
  min_height: "60" # cm
  max_height: "120" # cm
  c_screen_pin: D2
  c_tx_pin: D5
  c_rx_pin: D6
  k_screen_pin: D7
  k_tx_pin: D0
  k_rx_pin: D1
  friendly_name: "Mu Desk"
  name: "desk-1"

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  includes:
    - desk_height_sensor.h
    - desk_keypad.h
  on_boot:
    priority: -10
    then:
      - switch.turn_on: switch_wake_up

external_components:
  source: github://iMicknl/LoctekMotion_IoT
  components: [ loctekmotion_desk_height ]

logger:

esp8266:
  board: nodemcuv2

wifi:
  ssid: ""
  password: ""

  ap:
    ssid: "recover me ${friendly_name}"

captive_portal:

mqtt:
  broker: 192.168.1.3
  discover_ip: false
  discovery: false
  discovery_retain: false
  on_message:
    - topic: cmnd/desk/sit
      qos: 0
      then:
        - button.press: button_sit
    - topic: cmnd/desk/stand
      qos: 0
      then:
        - button.press: button_stand
    - topic: cmnd/desk/screen
      qos: 0
[... 304 lines cut ...]
            # direction up
            - cover.open: desk_cover
            - wait_until:
                lambda: |-
                  return id(desk_height).state  >= x - 1;
            - cover.stop: desk_cover
            - delay: 1s
            # move a little down again to adjust for momentum
            - uart.write: 
                id: desk_uart
                data: [0x9b, 0x06, 0x02, 0x02, 0x00, 0x0c, 0xa0, 0x9d] # down
          else:
            # direction down
            - cover.close: desk_cover
            - wait_until:
                lambda: |-
                  return id(desk_height).state <= x + 3;
            - cover.stop: desk_cover
            - delay: 1s
            # move a little up again to adjust for momentum
            - uart.write: 
                id: desk_uart
                data: [0x9b, 0x06, 0x02, 0x01, 0x00, 0xfc, 0xa0, 0x9d] # up
```
</details>

Nama, 2025-01-17:
I disconnected TX and bridged RX to the controller and everything works now.

So reading from keypad is not necessary.

Makr91, 2025-03-02:
> I disconnected TX and bridged RX to the controller and everything works now.
> 
> So reading from keypad is not necessary.

Now that ESPhome has deprecated custom components, do you by chance have an updated version of this?

Nama, 2025-03-02:
No, I didn't update.