iMicknl/LoctekMotion_IoT #81: HS11A-1C0 + CB38M2H(PB)-1 compatibility ?

M3te0r, 2024-03-17:
Hi, 
Do you know if the HS11A-1C0 + CB38M2H(PB)-1 combo is compatible?
[image]
[image]
[image]
Thanks!

M3te0r, 2024-03-17:
[image]
[image]

M3te0r, 2024-03-17:
Update : Tested out the HS13B-1 pins conf since they seemed to correspond with the control panel wiring, took a LOLIN S3 Mini  and a RJ45 connector, and it worked !

I had to adapt some parts of the config to match pins, min/mex height and the 3rd preset

Here is my yaml 

````
esphome:
  name: flexispot
  friendly_name: Flexispot 
  includes:
    - desk_height_sensor.h

esp32:
  board: lolin_s3_mini
  variant: ESP32S3
  framework:
    version: latest
    platform_version: 6.5.0
    type: arduino

# Enable logging
logger:
  #level: DEBUG
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: "[REDACTED]"

ota:
  password: "[REDACTED]"

uart:
  id: desk_uart
  baud_rate: 9600
  tx_pin: 43
  rx_pin: 44

sensor:
  - platform: wifi_signal
    name: "WiFi Signal"
    update_interval: 60s

  - platform: uptime
    name: Uptime

  - platform: custom
    lambda: |-
      auto desk_height_sensor = new DeskHeightSensor(id(desk_uart));
      App.register_component(desk_height_sensor);
      return {desk_height_sensor};
    sensors:
      id: "desk_height"
      name: Desk Height
      unit_of_measurement: cm
      accuracy_decimals: 1
      icon: "mdi:counter"

switch:
  - platform: gpio
    name: "Virtual Screen"
    pin:
      number: 33
      mode: OUTPUT
    restore_mode: ALWAYS_ON
    internal: true

  - platform: uart
    name: "Preset 1"
    id: switch_preset1
    icon: mdi:numeric-1-box
    data: [0x9b, 0x06, 0x02, 0x04, 0x00, 0xac, 0xa3, 0x9d]
    uart_id: desk_uart

  - platform: uart
    name: "Preset 2"
    id: switch_preset2
    icon: mdi:numeric-2-box
    data: [0x9b, 0x06, 0x02, 0x08, 0x00, 0xac, 0xa6, 0x9d]
    uart_id: desk_uart

  - platform: uart
    name: "Preset 3"
    id: switch_preset3
    icon: mdi:numeric-3-box
    data: [0x9b, 0x06, 0x02, 0x10, 0x00, 0xac, 0xac, 0x9d]
    uart_id: desk_uart

  - platform: uart
    name: "Stand" # Not available for all control panels
    id: switch_stand
    icon: mdi:human-handsup
    data: [0x9b, 0x06, 0x02, 0x10, 0x00, 0xac, 0xac, 0x9d]
    uart_id: desk_uart

  - platform: uart
    name: "Up"
    id: switch_up
    icon: mdi:arrow-up-bold
    data: [0x9b, 0x06, 0x02, 0x01, 0x00, 0xfc, 0xa0, 0x9d]
    uart_id: desk_uart
    internal: true

  - platform: uart
    name: "Down"
    id: switch_down
    icon: mdi:arrow-down-bold
    data: [0x9b, 0x06, 0x02, 0x02, 0x00, 0x0c, 0xa0, 0x9d]
    uart_id: desk_uart
    internal: true

  - platform: uart
    name: "M"
    id: switch_m
    icon: mdi:alpha-m-circle
    data: [0x9b, 0x06, 0x02, 0x20, 0x00, 0xac, 0xb8, 0x9d]
    uart_id: desk_uart

  - platform: uart
    name: "(wake up)" # Not available on all control panels
    id: switch_wake_up
    icon: mdi:gesture-tap-button
    data: [0x9b, 0x06, 0x02, 0x00, 0x00, 0x6c, 0xa1, 0x9d]
    uart_id: desk_uart

cover:
  - platform: template
    # icon: mdi:table-chair
    # icon: mdi-human-male-height-variant
    name: "Desk"
    assumed_state: true

    # Move desk up
    open_action:
      - while:
          condition:
            sensor.in_range:
              id: desk_height
              below: "126.9"
          then:
          - logger.log: "Executing up command"
          - switch.turn_on: switch_up
          - delay: 10ms
    
    # Move desk down
    close_action:
      - while:
          condition:
            sensor.in_range:
              id: desk_height
              above: "62.21"
          then:
          - logger.log: "Executing down command"
          - switch.turn_on: switch_down
          - delay: 10ms
    optimistic: true

````

Question is : 
I still want to use the keypad and there is only one RJ45 connector on the control block, the ESP32 can be powered on from the +5V, so could I just wire the ESP32 on the keypad PCB, right ?

M3te0r, 2024-03-23:
Update 2
Decided to buy 2 RJ45 female terminal connectors to handle the whole thing
[image]

ElCuboNegro, 2024-09-22:
I'm glad to see I'm not the only one triying to work with this specific interface.

I'm triying to make a "men in the middle" with an arduino mega, just to know how everything is moving (also, this is my first electronics experiment)

Paulomart, 2024-10-17:
Hey @M3te0r,

I have the same controller, but my desks stops when sending one down or up command (0x9b, 0x06, 0x02, 0x02, 0x00, 0x0c, 0xa0, 0x9d). When moving the desk it stops between movements, making the move unsteady. Do you also have the same issue?

Im using an `ESP8266 ESP-12F NodeMcu` board.

M3te0r, 2024-10-18:
Hi @Paulomart 
The up/down commands have the same behaviour as the physical up/down button, a single slighty hold push will wake up the terminal and send a single up/down command, a longer push hold will continuously send the commands until you release the button

To simulate that hold behaviour you have to loop through a cycle of wake up, up/down, that's the intent behind the `cover` template, take a look at the `open_action` you might have to fine tune the delay, don't go too short or it will spam the terminal and you'll need to reset the controller

Paulomart, 2024-10-20:
@M3te0r thanks I got it to work that way.

An issue I was facing: the sensor for the height was firing to often. I debounced the sensor and that seemed to help with smoother movement. 

Here is my final setup:

```yaml
api:
  encryption:
    key: "xxxxxxx"

ota:
  - platform: esphome
    password: "xxxxxxxx"

substitutions:
  min_height: "64" # cm
  max_height: "124" # cm
  screen_pin: D2
  tx_pin: D5
  rx_pin: D6
  friendly_name: "Office 1 Desk 1"
  name: "office-1-desk-1"

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  includes:
    - desk_height_sensor.h
  on_boot:
    priority: -10
    then:
      - switch.turn_on: switch_wake_up

external_components:
  source: github://iMicknl/LoctekMotion_IoT@v2
  components: [ loctekmotion_desk_height ]

logger:

esp8266:
  board: nodemcuv2

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    ssid: "recover me ${friendly_name}"

captive_portal:

uart:
  id: desk_uart
  baud_rate: 9600
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}

sensor:
  - platform: wifi_signal
    name: "WiFi Signal"
    update_interval: 60s

  - platform: uptime
    name: Uptime

  - platform: loctekmotion_desk_height
    id: "desk_height"
    name: Desk Height
    filters:
    - clamp:
        min_value: ${min_height}
        max_value: ${max_height}
        ignore_out_of_range: true
    - debounce: 10ms
    on_value_range:
      - below: ${min_height}
        then:
          - switch.turn_off: switch_down
      - above: ${max_height}
        then:
          - switch.turn_off: switch_up
    on_value:
      then:
        - component.update: set_desk_height

button:
  - platform: uart
    name: "Step Up"
    id: button_up
    icon: mdi:arrow-up-bold
    data: [0x9b, 0x06, 0x02, 0x01, 0x00, 0xfc, 0xa0, 0x9d]
    uart_id: desk_uart

  - platform: uart
    name: "Step Down"
    id: button_down
    icon: mdi:arrow-down-bold
    data: [0x9b, 0x06, 0x02, 0x02, 0x00, 0x0c, 0xa0, 0x9d]
    uart_id: desk_uart

switch:
  - platform: gpio
    name: "Virtual Screen"
    pin:
      number: ${screen_pin}
      mode: OUTPUT
    restore_mode: ALWAYS_ON
    internal: true

  - platform: uart
    name: "Up"
    id: switch_up
    icon: mdi:arrow-up-bold
    data: [0x9b, 0x06, 0x02, 0x01, 0x00, 0xfc, 0xa0, 0x9d]
    uart_id: desk_uart
    send_every: 1ms
    on_turn_on: 
      then:
        - switch.turn_off: switch_down
    internal: true

  - platform: uart
    name: "Down"
    id: switch_down
    icon: mdi:arrow-down-bold
    data: [0x9b, 0x06, 0x02, 0x02, 0x00, 0x0c, 0xa0, 0x9d]
    uart_id: desk_uart
    send_every: 1ms
    on_turn_on: 
      then:
        - switch.turn_off: switch_up
    internal: true

  - platform: uart
    name: "(wake up)" # Not available on all control panels
    id: switch_wake_up
    icon: mdi:gesture-tap-button
    data: [0x9b, 0x06, 0x02, 0x00, 0x00, 0x6c, 0xa1, 0x9d]
    uart_id: desk_uart
    internal: true

cover:
  - platform: template
    id: desk_cover
    icon: mdi:table-chair
    name: "Desk"
    assumed_state: true
    optimistic: true

    # Move desk up
    open_action:
      - if: 
          condition: 
            - switch.is_on: switch_down
          then: 
            - switch.turn_off: switch_down
            - delay: 500ms
      - switch.turn_on: switch_up

    # Move desk down
    close_action:
      - if: 
          condition: 
            - switch.is_on: switch_up
          then: 
            - switch.turn_off: switch_up
            - delay: 500ms
      - switch.turn_on: switch_down
    
    stop_action: 
      then:
        - switch.turn_off: switch_down
        - switch.turn_off: switch_up

number:
  - platform: template
    name: "Desk Height"
    id: set_desk_height
    min_value: ${min_height}
    max_value: ${max_height}
    icon: "mdi:counter"
    unit_of_measurement: "cm"
    device_class: "distance"
    step: 0.1
    lambda: !lambda |-
      return id(desk_height).state;
    set_action:
      - if:
          condition:
            - lambda: !lambda |-
                return x > id(desk_height).state;
          then:
            # direction up
            - cover.open: desk_cover
            - wait_until:
                lambda: |-
                  return id(desk_height).state  >= x - 1;
            - cover.stop: desk_cover
            - delay: 1s
            # move a little down again to adjust for speed
            - uart.write: 
                id: desk_uart
                data: [0x9b, 0x06, 0x02, 0x02, 0x00, 0x0c, 0xa0, 0x9d] # down
          else:
            # direction down
            - cover.close: desk_cover
            - wait_until:
                lambda: |-
                  return id(desk_height).state <= x + 3;
            - cover.stop: desk_cover
            - delay: 1s
            # move a little up again to adjust for speed
            - uart.write: 
                id: desk_uart
                data: [0x9b, 0x06, 0x02, 0x01, 0x00, 0xfc, 0xa0, 0x9d] # up

```

The-May, 2025-01-09:
Thanks for the extensive documentation @M3te0r!!!
I have the same PSU with my E.For.U® DT30 aka Ergofino DT30. (NOT PRO)
[image]

and i have another Button PCB but it seems to be similiar.
Color coding seems to be right/identical and since I have the same PSU i think it is similiar/same like your buttons.
front:
[image]
back:
[image]
[image]

PSU:
[image]

Buttons (not disassembled)
[image]

will test it in some time and report.

Nama, 2025-01-13:
I got my E5 and it came with HS36B-1 and CB38M2H(PB)-2, which seem identical to yours, @M3te0r.

Got it working with @Paulomart's config + MQTT on an ESP8266 NodeMCU and am trying to implement the keypad passthrough.
Had a hard time understanding your wiring table :D
But I guess every wire not listed can be passthough'd.

Whats the reason you didn't use the 5+v to power the ESP?

sinowine, 2025-11-12:
> Thanks for the extensive documentation @M3te0r!!!
>
> I have the same PSU with my E.For.U® DT30 aka Ergofino DT30. (NOT PRO)
>
>
>
> and i have another Button PCB but it seems to be similiar.
>
> Color coding [...]

Hey The-May, I also have the Ergofino DT30 with the CB38M2H(PB)-1 controller and the Loctek RF378A V1.5 20200505 control panel. Did you get it working? Specifically, did you figure out which pin is used for what?

sinowine, 2025-11-15:
> Yeah I got it working but not the passthrough-Version which then I ditched and reverted back to stock.

Do you still remember the pin assignment, or did you write it down and could you post it here? I would then test it again myself with the passthrough version.

The-May, 2025-11-15:
@sinowine i used default one in https://github.com/iMicknl/LoctekMotion_IoT?tab=readme-ov-file#pin-out iirc. (take that with a grain of salt!!). Im not responsible if you screw something up.
If I were you i would just take a RJ45 Cable plug it into the Desk and make sure (measure with multimeter) where GND and 5V is. If it is exactly like the documentation id just try it.

but in that time passthrough didnt work for my solution or was not existent (dont know if it exists now - i doubt not)

The reason I ditched it was that over some time my desk needed recalibration or pressing specific buttons to "unstuck" it and i couldnt unstuck it with the M-Button from the esp provided. This forced me to unplug it, replug the "dumb" buttons and do it, leaving me frustrated/annoyed and rethinking my decision to have such a solution.

Heck, I even created a "in between" solution but since recalibration didnt work It was not even  good like passthrough would be:
https://www.thingiverse.com/thing:7043452

but it does not change the fact that i could "unstuck" it, leaving me on the same issue like before.

lasharor, 2026-10-08:
I just want to confirm that the default solution seems to work fine. I just cut the cable and wired in the ESP32 inbetween. Now controllable by the ESP and the control panel.

lasharor, 2026-10-10:
[image]

For inspiration printed a very simple cover. Put the esp32 in there.

closed 2026-10-28 by iMicknl