iMicknl/LoctekMotion_IoT #86: [v2] compile errors for desk_height_sensor.cpp

iMicknl, 2024-06-07:
```
src/esphome/components/loctekmotion_desk_height/desk_height_sensor.cpp:12:16: error: 'byte' was not declared in this scope
 int hex_to_int(byte s)
                ^~~~
src/esphome/components/loctekmotion_desk_height/desk_height_sensor.cpp:63:17: error: 'byte' was not declared in this scope
 bool is_decimal(byte b)
                 ^~~~
src/esphome/components/loctekmotion_desk_height/desk_height_sensor.cpp: In member function 'virtual void esphome::loctekmotion_desk_height::DeskHeightSensor::loop()':
src/esphome/components/loctekmotion_desk_height/desk_height_sensor.cpp:73:9: error: 'byte' was not declared in this scope
         byte incomingByte = this->read();
         ^~~~
src/esphome/components/loctekmotion_desk_height/desk_height_sensor.cpp:77:13: error: 'incomingByte' was not declared in this scope
         if (incomingByte == 0x9b)
             ^~~~~~~~~~~~
src/esphome/components/loctekmotion_desk_height/desk_height_sensor.cpp:87:34: error: 'incomingByte' was not declared in this scope
             this->msg_len = (int)incomingByte;
                                  ^~~~~~~~~~~~
src/esphome/components/loctekmotion_desk_height/desk_height_sensor.cpp:93:30: error: 'incomingByte' was not declared in this scope
             this->msg_type = incomingByte;
                              ^~~~~~~~~~~~
src/esphome/components/loctekmotion_desk_height/desk_height_sensor.cpp:103:21: error: 'incomingByte' was not declared in this scope
                 if (incomingByte == 0)
                     ^~~~~~~~~~~~
src/esphome/components/loctekmotion_desk_height/desk_height_sensor.cpp:108:49: error: 'esphome::loctekmotion_desk_height::hex_to_int' cannot be used as a function
                 else if (hex_to_int(incomingByte) == 0)
                                                 ^
src/esphome/components/loctekmotion_desk_height/desk_height_sensor.cpp:135:58: error: 'esphome::loctekmotion_desk_height::hex_to_int' cannot be used as a function
                 int height1 = hex_to_int(this->history[1]) * 100;
                                                          ^
src/esphome/components/loctekmotion_desk_height/desk_height_sensor.cpp:136:58: error: 'esphome::loctekmotion_desk_height::hex_to_int' cannot be used as a function
                 int height2 = hex_to_int(this->history[0]) * 10;
                                                          ^
src/esphome/components/loctekmotion_desk_height/desk_height_sensor.cpp:137:42: error: 'incomingByte' was not declared in this scope
                 int height3 = hex_to_int(incomingByte);
                                          ^~~~~~~~~~~~
src/esphome/components/loctekmotion_desk_height/desk_height_sensor.cpp:137:54: error: 'esphome::loctekmotion_desk_height::hex_to_int' cannot be used as a function
                 int height3 = hex_to_int(incomingByte);
                                                      ^
src/esphome/components/loctekmotion_desk_height/desk_height_sensor.cpp:144:52: error: 'esphome::loctekmotion_desk_height::is_decimal' cannot be used as a function
                     if (is_decimal(this->history[0]))
                                                    ^
src/esphome/components/loctekmotion_desk_height/desk_height_sensor.cpp:159:28: error: 'incomingByte' was not declared in this scope
         this->history[0] = incomingByte;
                            ^~~~~~~~~~~~
Compiling .pioenvs/office-desk-flexispot-ek5/src/esphome/components/mdns/mdns_esp32.cpp.o
*** [.pioenvs/office-desk-flexispot-ek5/src/esphome/components/loctekmotion_desk_height/desk_height_sensor.cpp.o] Error 1
========================= [FAILED] Took 25.39 seconds =========================
```

iMicknl, 2024-06-07:
@jamesmyatt did you face similar issues on the latest version? I will look into this as well, but not sure if I am doing anything wrong here.

iMicknl, 2024-06-07:
Including `#include <cstdint>` and changing `byte` to `std::uint8_t` seems to fix the issue. Haven't tested it on device yet.

jamesmyatt, 2024-06-07:
Works for me with the ESPHome 2024.5.5 addon. Here's my config exactly (except for the device name): https://gist.github.com/jamesmyatt/570aa3e886ca748c23d572bfb324d90e

iMicknl, 2024-06-07:
@jamesmyatt interesting, for me in 2024.5.5 this doesn't work. Both locally in a just created devcontainer as on my HA Core OS. Both run Python 3.12 and esphome==2024.5.5.

Before the change to custom component, did the sensor updates work? Or did you face the same issue?

jamesmyatt, 2024-06-08:
I think the sensor updates was the same before. I need to fire up my logic analyser to see what's going on with the serial interface.

I'll also take try compiling in a ESPHome dev container later, but I've been using the official ESPHome add-on with HA OS x86 VM.

iMicknl, 2024-06-08:
@jamesmyatt can you perhaps try my latest changes in branch 2.3 when you have time? 

I added some CI/CD to spot these issues more easily, and it fails on the same error on CI as on my local machine. See https://github.com/iMicknl/LoctekMotion_IoT/actions/runs/9428679879/job/25974279658

jamesmyatt, 2024-06-09:
Also compiles fine for me using the `esphome/esphome:2024.5.5` image as a devcontainer, for both v2 and v2.3 branches.

I think this is the associated dockerfile: https://github.com/esphome/esphome/blob/dev/docker/Dockerfile. It uses Python 3.11.2 rather than 3.12.

iMicknl, 2024-06-10:
[image]

@jamesmyatt I am really lost on why this wouldn't compile for me, while it does for you. Have you tried v2.3?

jamesmyatt, 2024-06-10:
Which YAML file are you using?

iMicknl, 2024-06-11:
https://github.com/iMicknl/LoctekMotion_IoT/blob/v2.3/packages/office-desk-esp32.yaml

jamesmyatt, 2024-06-11:
That fails for me in the ESPHome addon

jamesmyatt, 2024-06-11:
OK. Your config works for me when I add an MQTT component. This must be including some extra headers that are defining `byte` etc.

```

# Enable MQTT
mqtt:
  broker: !secret mqtt_broker
  port: !secret mqtt_port
  username: !secret mqtt_username
  password: !secret mqtt_password
  discovery: false
  topic_prefix: "esphome/${device_name}"
  ```

iMicknl, 2024-06-14:
@jamesmyatt best thing would be to change the definition in the files. Would you mind testing https://github.com/iMicknl/LoctekMotion_IoT/tree/v2.3 or https://github.com/iMicknl/LoctekMotion_IoT/tree/2.4? Both have a different implementation. 2.4 is cleaner in my opinion and seems to work well for my desk.

jamesmyatt, 2024-06-15:
I think the right type here is `uint8_t` like here: https://github.com/esphome/esphome/blob/2fc43fa9c71e7f1a5785efffee787033366ca188/esphome/components/uart/uart.h#L29

In fact, it looks like it's more correct to use the `read_byte` method, rather than the `read` one to get the bytes.

jamesmyatt, 2024-06-15:
I have a PR for this

jamesmyatt, 2024-06-15:
See #89

iMicknl, 2024-06-15:
Thanks @jamesmyatt! By the way, are you on Discord as well? Happy to add you as contributor on this repo, if you want to.

closed 2024-06-15 by iMicknl