iMicknl/LoctekMotion_IoT #92: Flexispot E8: Wiring for ESP32

lufetico, 2024-07-26:
Hi there,

I have assumed, I believe correctly, that my desktop Flexispot E8 uses the same controller as the E7, the HS13B-1.

I have uploaded the firmware to an ESP32 for use in ESPHome but I can't get my head around the wiring.

I am using this board:

[image]

I have gone through all the documentation and I can't find where the table is with the wiring from HS13B-1 to ESP32.

I can only find the wiring for ESP8266 but there is no correspondence of the pins of that chip with mine. 

Thanks!

Regards,

lufetico, 2024-07-26:
Confirmed that the controller is an HS13B-1.

lufetico, 2024-07-26:
I have the following wiring:

[image]

And config in ESP32:

[image]

I can change the table elevation from Home Assistant but the other functions do not work. Especially the table height is not updated in the ESP32 when it is changed from the physical panel.

iMicknl, 2024-07-28:
Have you already tried the new pin-out and YAML from https://github.com/iMicknl/LoctekMotion_IoT/issues/67?

lufetico, 2024-07-29:
I will test it and then get back to you. Thanks!