iMicknl/LoctekMotion_IoT #94: Flexisport E5-A HCB210C-2

ijohnston1911, 2024-09-12:
I ordered a E5-A thinking I would get an E5 controller.  There is no RJ45 port, only a USB-A and a USB-C on the side .  These are both mentioned in the manual as being there for charging phones.  There is an unmentioned USB-A port that had a cover on the back of the controller that seemed to be a likely contender, but after tearing it down down and tracing continuity across the pcb this seems to be for SWD.

The keypad is connected to board with all the connectors via an 11 pin  ribbon cable.  I tried probing it and annotating an image but it turned out that the ribbon cable has more conductors on the other side and I wasn't mapping them properly. I plan to try again tomorrow to more effectively map the pinout and see if I can find a UART TX / RX pin.

Here's some pictures of the board I'm working with if anyone has any recommendations

[image]

[image]

[image]

Its an STM32G030 C8T6 microcontroller 

and the datasheet im going off of is here https://www.st.com/resource/en/datasheet/stm32g030c6.pdf

jamesmyatt, 2024-09-12:
What do CN1, etc connect to?

ijohnston1911, 2024-09-19:
After some poking around it looks like there's no serial interface to inject commands into.  

For control, I ended up soldering to the test pads for the buttons and connecting to a Raspberry Pi GPIO. By pulling the GPIO pins to ground I can simulate a button press.

To get the current height of the desk, ultimately the best option was reading the pins from the 7 segment display.  With the casing open and looking at the pcb with the STM32 from behind (as in the first picture) , Pin 1 is the top left most pin and the number increases in a clockwise configuration:

| | | | | | |
|---|---|---|---|---|---|
|1  |2  |3  |4 |5|  |
|12|11|10|9|8|7|

Pin 6 is not exposed.  Pins 12, 9, and 8 are the common pins for each digit (tens, ones, tenths respectively).  By using these three pins as interrupts and reading the state of the rest I was able to parse out the reported height in a way that consistently matched the display.  Very occasionally I run into a timing issue and its not able to decode the signals correctly using this method, but it is infrequent and workable for my purposes. The segment lettering that corresponds to the graphic in the readme is as follows:

| | | | | | |
|---|---|---|---|---|---|
|E  |D  |DP  |C |G|  |
|12|A|F|9|8|B|

Investigating the ribbon connector I've found pins for 

- Down Enable
- Down Motor 1 PWM
- Down Motor 2 PWM
- Up Enable
- Up Motor 1 PWM
- Up Motor 2 PWM
- Motor 1 Hall 1
- Motor 1 Hall 2
- Motor 2 Hall 1
- Motor 2 Hall 2

but I opted not to pursue the ribbon cable further because it looks like everything I can tap into there is going to be relative positioning and the STM32 is storing the current position in some form of persistent memory.  There's a possibility it could be accessed using the SWD interface (rear usb) but I lack the experience to access the memory registers through that interface.

I hope this is helpful for anyone that ends up with an E5-A and is hoping to control it externally.