iMicknl/LoctekMotion_IoT #95: Desk Height

wernerhp, 2024-09-30:
The desk height component has some glitches.

When setting the height from 72.5cm to 105cm it starts but then stops shortly after with with a height value of 800cm (I suspect this hsould be 80.0cm).
Similarly, when it lowers from 105cm to 72.5, it stops shortly after with a height value of 10cm (I suspect this should be 100.0cm)

[image]
[image]
[image]

https://github.com/user-attachments/assets/eba648d9-71a5-4aa3-b842-8471fe5c6cf0

iMicknl, 2024-10-18:
Which version do you use? We just published a new version of this component, see the README. 

Number entity will still overshoot, but in general the code is better.

wernerhp, 2024-10-18:
I tried `main` and `v2`

```
source: github://iMicknl/LoctekMotion_IoT@v2
  components: [ loctekmotion_desk_height ]
```

wernerhp, 2024-10-18:
[image]

I'd have expected the cponent to be updated. That's where the problem appeared to have been

iMicknl, 2024-10-18:
@wernerhp I would need to investigate if I face the same issue, or if your device is different..

wernerhp, 2024-10-22:
Have a look at this height decoder 
https://www.mikrocontroller.net/attachment/452474/Schreibtisch_Dekoder.xlsx 

Maybe you could add a test to check that the examples form the sheet are correctly decoded.