iMicknl/LoctekMotion_IoT #103: [Question] known Flexispot Ethernet desks

tuxtux59, 2024-11-14:
Hi,

I land on your repo from **Everything Smart Home** video. 
I think about buying my first standing desk and I would like to pick one compatible with your solution.
What are the known compatible desks at the moment ? I'm looking not for the cheapest one but not buying the expensive one (ex E7).

Thanks

jamesmyatt, 2024-11-15:
"RJ45" not "ethernet", but my guess is that the E7 is the cheapest current model with 2 RJ45 ports. Cheaper ones often have either 1 RJ45 (which requires passthrough), or have integrated keypad/controllers with no ports.

tuxtux59, 2024-11-15:
yeah my bad @jamesmyatt  on Ethernet <> RJ45 ;) .
I saw one of flexispot desk that uses some two wires interface like this https://youtu.be/r1hWZ6SbrlA?si=PpqtWejw3F6o6TjW&t=75 but maybe I'm wrong and It's here the power supply ?

jamesmyatt, 2024-11-15:
On that controller, "HS" = handset, so it has 1 RJ45. Then M1,M2 = motor 1, motor 2. The last 2 are for power (I assume).

meistermolli, 2024-12-16:
Hi, Flexispot E2B (299Euro) has 2 RJ45. I will test it soon