iMicknl/LoctekMotion_IoT #113: cb38m2l - Down button not working in HASS

mikedhanson, 2025-01-03:
[image]

Any ideas? 

I have an wemos d1 mini and followed the following wiring https://github.com/iMicknl/LoctekMotion_IoT/tree/main?tab=readme-ov-file#hs13b-1 

```yaml
substitutions:
  device_name: flexispot-e7
  name: flexispot-e7
  min_height: "73.5" # cm
  max_height: "123" # cm
  tx_pin: D5 # = GPIO14 blue white 5
  rx_pin: D6 # = GPIO12 green 6
  screen_pin: D2 # = GPIO4 
  encryption_key: "="

external_components:
  source: github://iMicknl/LoctekMotion_IoT
  components: [ loctekmotion_desk_height ]

esphome:
  name: ${name}
  friendly_name: ${device_name}
  comment: Used to control your ${device_name} standing desk via Home Assistant.

  # Wake Desk by sending the "M" command
  # This will pull the current height after boot
  on_boot:
    priority: -10
    then:
      - button.press: button_m

esp8266:
  board: d1_mini

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: ${encryption_key}

ota:
  - platform: esphome
    password: "c7123d9ea62a0eaf6f056b60d618b3ce"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Flexispot-E7 Fallback Hotspot"
    password: "pcGBr1liQM69"

captive_portal:

uart:
  - id: desk_uart
    baud_rate: 9600
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}

sensor:
  - platform: wifi_signal
    name: "WiFi Signal"
    update_interval: 60s

  - platform: uptime
    name: Uptime

  - platform: loctekmotion_desk_height
    id: "desk_height"
    name: Desk Height
    on_value_range:
    - below: ${min_height}
      then:
        - switch.turn_off: switch_down
    - above: ${max_height}
      then:
        - switch.turn_off: switch_up
    on_value:
      then:
        - cover.template.publish:
            id: desk_cover
            position: !lambda |-
                // The sensor outputs values from min_height (cm) to max_height (cm)
                // We need to translate this to 0 - 1 scale.
                float position = (float(x) - float(${min_height})) / (float(${max_height}) - float(${min_height}));
                return position;
        - component.update: set_desk_height

switch:
  - platform: gpio
    name: "Virtual Screen" # PIN20
    pin:
      number: ${screen_pin}
      mode: OUTPUT
    restore_mode: ALWAYS_ON
    entity_category: "config"
    internal: true

  - platform: uart
    name: "Up"
    id: switch_up
    icon: mdi:arrow-up-bold
    data: [0x9b, 0x06, 0x02, 0x01, 0x00, 0xfc, 0xa0, 0x9d]
    uart_id: desk_uart
    send_every: 108ms
    internal: true

  - platform: uart
    name: "Down"
    id: switch_down
    icon: mdi:arrow-down-bold
    data: [0x9b, 0x06, 0x02, 0x02, 0x00, 0x0c, 0xa0, 0x9d]
    uart_id: desk_uart
    send_every: 108ms
    internal: true

  - platform: uart
    name: "Alarm off"
    id: switch_alarm
    icon: mdi:alarm
    data: [0x9b, 0x06, 0x02, 0x40, 0x00, 0xAC, 0x90, 0x9d]
    uart_id: desk_uart
    send_every: 108ms
    on_turn_on:
      - delay: 3000ms
      - switch.turn_off: switch_alarm
    entity_category: "config"

  - platform: uart
    name: "Child Lock"
    id: switch_child_lock
    icon: mdi:account-lock
    data: [0x9b, 0x06, 0x02, 0x20, 0x00, 0xac, 0xb8, 0x9d]
    uart_id: desk_uart
    send_every: 108ms
    on_turn_on:
      - delay: 5000ms
      - switch.turn_off: switch_child_lock
    entity_category: "config"

button:
  - platform: template
    name: "Preset 1"
    icon: mdi:numeric-1-box
    on_press:
      - uart.write:
          id: desk_uart
          data: [0x9b, 0x06, 0x02, 0x04, 0x00, 0xac, 0xa3, 0x9d]

  - platform: template
    name: "Preset 2"
    icon: mdi:numeric-2-box
    on_press:
      - uart.write:
          id: desk_uart
          data: [0x9b, 0x06, 0x02, 0x08, 0x00, 0xac, 0xa6, 0x9d]

  - platform: template
    name: "Sit" # Preset 3 on some control panels
    icon: mdi:chair-rolling
    on_press:
      - uart.write:
          id: desk_uart
          data: [0x9b, 0x06, 0x02, 0x00, 0x01, 0xac, 0x60, 0x9d]

  - platform: template
    name: "Stand" # Preset 4 on some control panels
    icon: mdi:human-handsup
    on_press:
      - uart.write:
          id: desk_uart
          data: [0x9b, 0x06, 0x02, 0x10, 0x00, 0xac, 0xac, 0x9d]

  - platform: template
    name: "Memory"
    id: button_m
    icon: mdi:alpha-m-box
    entity_category: "config"
    on_press:
      - uart.write:
          id: desk_uart
          data: [0x9b, 0x06, 0x02, 0x20, 0x00, 0xac, 0xb8, 0x9d]

  - platform: template
    name: "Wake Screen"
    id: button_wake_screen
    icon: mdi:gesture-tap-button
    entity_category: "config"
    on_press:
      - uart.write:
          id: desk_uart
          data: [0x9b, 0x06, 0x02, 0x00, 0x00, 0x6c, 0xa1, 0x9d]

  - platform: template
    name: "Alarm"
    id: button_alarm
    icon: mdi:alarm
    on_press:
      - uart.write:
          id: desk_uart
          data: [0x9b, 0x06, 0x02, 0x40, 0x00, 0xAC, 0x90, 0x9d]

  - platform: restart
    name: "Restart"
    entity_category: "config"

cover:
  - platform: template
    id: "desk_cover"
    icon: mdi:desk # or mdi:human-male-height-variant
    name: "Desk"
    device_class: blind # makes it easier to integrate with Google/Alexa
    has_position: true
    position_action:
      - if:
          condition:
            - lambda: !lambda |-
                return pos > id(desk_cover).position;
          then:
            - cover.open: desk_cover
            - wait_until:
                lambda: |-
                  return id(desk_cover).position  >= pos;
            - cover.stop: desk_cover
          else:
            - cover.close: desk_cover
            - wait_until:
                lambda: |-
                  return id(desk_cover).position <= pos;
            - cover.stop: desk_cover
    stop_action:
      - switch.turn_off: switch_up
      - switch.turn_off: switch_down
    open_action:
      - switch.turn_off: switch_down
      - switch.turn_on: switch_up
    close_action:
      - switch.turn_off: switch_up
      - switch.turn_on: switch_down
    optimistic: false

number:
  - platform: template
    name: "Desk Height"
    id: set_desk_height
    min_value: ${min_height}
    max_value: ${max_height}
    icon: "mdi:counter"
    unit_of_measurement: "cm"
    device_class: "distance"
    step: 0.1
    lambda: !lambda |-
      return id(desk_height).state;
    set_action:
      - if:
          condition:
            - lambda: !lambda |-
                return x > id(desk_height).state;
          then:
            - cover.open: desk_cover
            - wait_until:
                lambda: |-
                  return id(desk_height).state  >= x;
            - cover.stop: desk_cover
          else:
            - cover.close: desk_cover
            - wait_until:
                lambda: |-
                  return id(desk_height).state <= x;
            - cover.stop: desk_cover
```

iMicknl, 2025-01-06:
The issue here is that your desk height is being read as 27.9cm. This is lower than 73.5 and thus you can't go down. Not sure why this is not read correctly.

You can remove internal: true for the up/down buttons as a temporary fix.

mikedhanson, 2025-01-14:
Commenting out internal: true allowed me to use the up and down feature.

iMicknl, 2025-02-21:
For my understanding does your desk use cm? I am not sure why we don't detect the right number. Is the desk height always wrong in Home Assistant?

JS876, 2025-04-05:
> For my understanding does your desk use cm? I am not sure why we don't detect the right number. Is the desk height always wrong in Home Assistant?

@iMicknl  just ran into this issue as well with my E7 Pros. The value is being read in inches (which matches the display), but the esphome config expects centimeters. Attached the device logs if that helps. It covers device boot up and some manual height adjustments.

- Display/Control Panel - Model HS13B-1
- Control Box - CB38M2L(IB)-1

I also noticed that the screen wake functionality doesn't work, but I need to try the other pin outs. 

[image]

If it helps, attached are some logs with uart debug enabled. It covers the device boot up and some manual adjustments of height: 

[logs-uart-debug-enabled.txt](https://github.com/user-attachments/files/19613447/logs-uart-debug-enabled.txt)

iMicknl, 2025-04-05:
Is the number correct or is the number also not correct? Have you tried changing the unit of measurement to inches? 

https://github.com/iMicknl/LoctekMotion_IoT/blob/dbcc514481ad28bbd9928718fb08ef8aa094e2ad/packages/office-desk-esp32.yaml#L254

Short term fix can be to remove `internal: true` as the up/down buttons (id is `switch_up`, `switch_down`).

JS876, 2025-04-05:
> unit_of_measurement: "cm"

Yes, I did. The returned value is incorrect when the unit of measure is changed. 

In this case it returns 11.2" inches instead of 28.5". This looks to be due to HA converting the value when the unit is changed.

Manually overriding the unit of measure in HA to "cm" displays the value of 28.5". I'll take another look later as well.

[image]

ianmtaylor1, 2026-02-14:
> Is the number correct or is the number also not correct? Have you tried changing the unit of measurement to inches?

I had the same problem, and setting `unit_of_measure: "in"` under **both** the `number:` and the `sensor:` solved it for me.

```
sensor:
  [...]
  - platform: loctekmotion_desk_height
    id: "desk_height"
    name: Desk height
    unit_of_measurement: "in" # <----- THIS
    [...]

[...]

number:
  - platform: template
    name: "Desk height"
    id: set_desk_height
    min_value: ${min_height}
    max_value: ${max_height}
    icon: "mdi:counter"
    unit_of_measurement: "in" # <----- THIS
    [...]
```

I also had to set the minimum and maximum height to the appropriate values in inches.

iMicknl, 2026-02-21:
@ianmtaylor1 for my understanding, did you set the max height to the value in inch? I will make some changes to the YAML to incorporate your changes.

Can you verify the changes in https://github.com/iMicknl/LoctekMotion_IoT/pull/180/changes? 👍🏻

closed 2026-10-28 by iMicknl