iMicknl/LoctekMotion_IoT #123: Desk Height not updated - Flexispot E8 Desk

nsleigh, 2025-04-02:
I am using this code with Flexispot E8 Desk and the function all work ok but the desk height is not always updated - not sure if this is specific to the E8 or a generic issue.

Working scenarios:

- Send Up or Down and then stop
- Send the memory command
- On initial boot (which calls the memory command)

Not working scenarios:

- Manually moving the desk
- Sending the Sit or Stand commands

In the debug log I can see it is sending the height every 60s but it is often wrong:
[10:10:37][D][number:012]: 'Desk Height': Sending state 69.000000

Is anyone else seeing this? Any thoughts on a fix?

nsleigh, 2025-08-17:
@pschmitt thanks, I'll give that a try.

lufetico, 2026-03-19:
> I am using this code with Flexispot E8 Desk and the function all work ok but the desk height is not always updated - not sure if this is specific to the E8 or a generic issue.
>
> Working scenarios:
>
> * [...]

Are you using ESPHome with ESP32? Could you provide the PIN mapping and YAML code you're using? I'm having trouble understanding the information I'm finding in this repository.