iMicknl/LoctekMotion_IoT #125: NO ISSUE: Control panel HS11B is working

pgStevie, 2025-04-18:
Just want to say: HS11B ist woring with HA and ESPHome.

Another question: Wouldn't it be nice to gather all the working / not working control panels, controller boxes with there wiring schemata and required version of the component in a central place?

iMicknl, 2025-04-18:
We actually had this in the past, but there are too many combinations of control panels + control boxes out there nowadays. It would be easier to document the ones that are not supported (yet) perhaps. In essence, every controller that has at least an RJ45 port can be supported.

How would such an overview look like? I am open to contributions, but my main challenge is that it needs to be easy to read.

pgStevie, 2025-04-18:
Understandable point of view! I've purchased my desk last week. Due to that I'm also new to this component. Didn't know that there are so many different combinations of boxes and controllers in the market. 

I was thinking about a table with a few columns for controller boxes and rows for the control panels. So it would be possible to check the working combinations with crosses. This would be easy to read.

Yes...this table would lack wiring infos and also required version of the component but for the first investigation, it could be helpful.

Anyway...thanks for the good work on this component.

iMicknl, 2025-04-18:
In the past we also had different wiring per control panel, but we found out that these wirings are universal and that the control box is able to adapt to different wiring schemes. The wiring currently published should work for the majority of the boxes. :)

pgStevie, 2025-04-18:
Hmm...it seems, live became easier by time! ;-)

closed 2026-10-28 by iMicknl